RoboJackets/robocup-software #441: Investigate toolchain for ATTiny 44A

justbuchanan, 2015-09-10:
This is the chip we'll be using for the kicker board.  We need figure out how to setup the toolchain for compiling code for this architecture.  Hopefully we can download a cross-compiler rather than compiling one.

cc @ryanstrat @jjones646 @guyfleeman

jjones646, 2015-09-10:
I just wanted to note that the issue number for this issue is really throwing me off.

On a more related note, the 44A is officially supported on the avr-gcc compiler. I've been able to do everything after `sudo apt-get install gcc-avr binutils-avr avr-libc`.

justbuchanan, 2015-09-10:
Haha can't believe I didn't even notice that.  Sounds good though, I'm glad we won't have to compile any compilers.

closed 2015-09-10 by justbuchanan