RoboJackets/robocup-software #445: Builds timeout on circleci

justbuchanan, 2015-09-10:
I've noticed a couple builds failing because circleci times out running a command.  See this build for example:

https://circleci.com/gh/RoboJackets/robocup-software/429

jgkamat, 2015-09-10:
On my system the `TargetVelPathPlannerTest.run` test seems to be infinite-looping. Could you double check that test?
Does gtest have something to kill tests after ~1min so they don't stop the entire build?

jgkamat, 2015-09-10:
Whats weird is that we don't see the output of the failing test until circle kills it. This could be because circle/docker gives  us a non-interactive shell? Not sure about this.

jgkamat, 2015-09-10:
Looks like this is failing as well on `justin/escape-obstacles-planner` as well. da81c8b0e9bba4439b581de93ceb288c75a8f808 seems to be the first bad commit

justbuchanan, 2015-09-10:
Nice, thanks for finding that!  The strange thing is that the unit tests worked fine on Arch and OS X, but not Ubuntu.  This was fixed in 6d79e2633f8e6e8d53f7081faeee0ccb0123b21f

closed 2015-09-10 by justbuchanan

jgkamat, 2015-09-10:
[looks like gtest dosen't have timeout functionality](https://code.google.com/p/googletest/issues/detail?id=348). We'll probably just have to live without it or find a way to see if no output is being generated.

justbuchanan, 2015-09-10:
Yeah it'd be nice if they had it, but I think it'll be pretty rare that we have a test freeze like that.  Especially rare that it'll freeze on one system, but not another.