RoboJackets/robocup-software #461: Add .gitmodules file to setup script checksum for circleci

justbuchanan, 2015-09-30:
Looks like changes to submodules are breaking some ci builds because the base image isn't being rebuilt.  This is the case for the ninja branch at the moment.

jgkamat, 2015-09-30:
This is solved by #460, I'll merge it later today (you could pull from it into the ninja branch). Keep in mind though that the [gitmodules file](https://github.com/RoboJackets/robocup-software/blob/master/.gitmodules) dosent change when submodules are *updated*, (no sha in the file). I don't know where the actual sha is stored though.

justbuchanan, 2015-09-30:
Cool, sounds good.  The SHAs are stored at the path of the submodule itself (some sort of weird file voodoo).

PR #460 (2015-10-01): Switch over to DoCIF
Now our Docker based CI is controlled from a singular config file, and is mostly portable to other platforms. I'm not 100% sure deployment works though, but other than that I've tested the things we personally use and they look good!

Closes #461
Lets not merge this until after the meeting earliest...
closed 2015-10-01 by jgkamat