RoboJackets/robocup-software #477: make robot2015-prog doesn't always work

justbuchanan, 2015-10-25:
It seems to only work if the mbed is mounted to the filesystem, the old script worked even when it wasn't mounted.

justbuchanan, 2015-10-25:
Note: this is on the `robot2015-firmware` branch, the version on master works fine.

justbuchanan, 2015-10-25:
@jjones646, the version of the programming script on the master branch seems to work fine as far as mounting and unmounting.  What all does the new one on the firmware branch add?

jjones646, 2015-10-26:
The one on the firmware branch searches for mbeds using the `blkid` command I believe, passing it the mbed label of the partition thinking it would help for something...can't remember exactly. But I'm almost positive it didn't fix whatever I was aiming for at the time. And sometimes the mbed likes to forget the partition label that was given to it when formatted...so yeah, that should definitely be changed back.

I did have the script in master fail quite a few times. Problems usually came up when having multiple mbeds connected which was quite often when I was doing radio things with 2 mbeds at once. Won't really be a problem down the road, but all the above escalated because it became very working with 2 mbeds for any extended period of time.

While I'm at it, here's where that python file fits in: It was started to basically have a reliable way of doing more test/debug type things with the mbed. It uses the pyOCD library (it's legit!) for loading binaries straight to flash, resetting, and recording anything the mbed sends back over the serial port during startup. That was used for running through multiple test binaries automatically and determining if it was a pass/fail result in hardware. All that is very reliable, but it does take quite some time to load it on the mbed, and that won't actually save it on the flash storage for later use.
I added copying the file to the mbed to the python script as it grew. It uses the copy command for that. That may be where things aren't going good actually. The copyfileobj function may be worth a shot, or just taking that part out all together. Having pyserial and pyOCD is what that script was really meant to be for - not moving files around.

justbuchanan, 2015-10-27:
Ok, that all makes more sense now.  I remember messing with pyOCD a while ago and it seemed pretty sweet.

I didn't have issues before with the unmounting and mounting with the old script, but maybe that was an ubuntu-specific thing (I've got a different linux distro that doesn't automount stuff).  I'll take a look at it and maybe pull some of the mounting code from the old script.

justbuchanan, 2015-11-04:
@jjones646, can you try out the `robot2015-prog.sh` script on `master` and see if it works for you?  It works great for me and supports multiple mbeds.  It copies the file if it's different, then reboots all mbeds.  I can't get your mbed-prgm.sh script to work unless I manually mount the mbed and even then it only works for one at a time.

jjones646, 2015-11-04:
Yeah, the one on `master` is the one that should be used by default. The python based one on `robot2015-firmware` should really only be used for the really low level debugging stuff. I'm pretty sure all the times when the script on `master` failed for me, I had a screen sessions open for each mbed - something that doesn't really need to be considered by the script.
In the `robot2015-firmware` branch, the script that is used for all of the mbed targets is set at [firmware/CMakeLists.txt:25](https://github.com/RoboJackets/robocup-software/blob/robot2015-firmware/firmware/CMakeLists.txt).

commit efc477f8 (2015-11-04): revert to old mbed copy script. closes #477

this one supports multilple mbeds being connected at once and
also doesn't require them to already be mounted
closed 2015-11-04 by justbuchanan