RoboJackets/robocup-software #481: Style checking failing when force pushing due to  being out of date.

jgkamat, 2015-10-27:


jgkamat, 2015-11-04:
I don't think this is an issue any more. We'll reopen if we see it again.

jgkamat, 2015-11-11:
This issue is present again on #499

justbuchanan, 2015-11-12:
@jgkamat, do you think we can fix this by adding `git fetch origin master:master` to the `SETUP_COMMAND` in `config.docif`?  The git fetch command will pull the latest master branch without checking it out.

jgkamat, 2015-11-12:
Oh I didn't know we could just do that one command, I'll go ahead and add that.

justbuchanan, 2015-11-12:
Looks like your fix is failing because of SSH issues.  I SSHd into circle and it looks like they're cloning from github with the `git@` ssh url, which requires an SSH key.  They have one at `~/.ssh/id_circleci_github`, but that isn't visible in docker.  Looks like the easiest fix would be to forward the whole `.ssh` folder into Docker if that's possible.

jgkamat, 2015-11-12:
Oh, nice, thanks for debugging that! I'll just forward that inside

jgkamat, 2015-11-12:
circleci has direct links in their ~/.ssh/config, I added a sed to swap those out for relative links

justbuchanan, 2015-11-12:
Sure thing, that should work.  Can't remember quite how circle has it setup, but you might be able to just copy their key and avoid any potential future breakages if they change their ssh config.

Another thing we should check on once that's fixed is the version of `clang-format` in our image.  We'll probably have to keep it as is since it'll match what the Ubuntu users have, it's just a little unfortunate that it's a few versions behind up-to-date distros (Arch, Debian, OS X).

commit dfdb36bd (2015-11-12): Try to fix style issues when running old CI cache
Closes #481
commit 080d3d04 (2016-08-02): Try to fix style issues when running old CI cache
Closes #481
commit 52356479 (2016-08-02): Try to fix style issues when running old CI cache
Closes #481

Former-commit-id: dfdb36bd0b995a50f2217efb7da5e7b30f078866
closed 2015-11-12 by jgkamat