RoboJackets/robocup-software #483: Send a packet with the new radio setup

justbuchanan, 2015-10-28:
@jjones646 has it very close, but there are a few things that may be misconfigured, preventing a successful tx/rx.

The [CC debugger](http://www.ti.com/tool/cc-debugger) program should help with this.

jjones646, 2015-10-28:
For reference, [figure 9 in this pdf](http://www.ti.com/lit/ug/swru197h/swru197h.pdf) shows the pinout for the connection I used on those breakout cables.

justbuchanan, 2015-11-03:
@jjones646, I noticed in the `CC1201` class, some register values are converted using the `twos_complement()` function before being written over SPI - how does this work?  If we're dealing with positive numbers, aren't they already valid in 2's complement?

jjones646, 2015-11-04:
It's just a quick way to force the compiler to flip the sign for an unsigned, fixed width type. The same type is returned that gets passed to the function. So the compiler knows how to arrange the bits, and all of the compiler warnings go away for any weird type casting that would be done otherwise.

justbuchanan, 2015-11-04:
So is the idea here that we're trying to pass a negative value to the cc1201 and we're circumventing the fact that the type of the variable is `uint8_t` in our code?

justbuchanan, 2015-11-09:
@jjones646, I exported a new radio config from SmartRF Studio and set it up in the firmware.  The logs show that the `CC1201.freq()` method is returning 910MHz, while my config has it set to 916.  Any ideas why this would happen?

justbuchanan, 2015-11-09:
I don't fully understand what's going on with the `SAFC` strobe and the `FREQOFF1` register.

justbuchanan, 2015-11-19:
It works!!!  I sent a packet from an MBED using `radio test-tx` to another 1201 in the cc debugger.  The CRC passed and the RSSI was -66.

justbuchanan, 2015-11-19:
For future reference, here are the [golden configs](https://github.com/RoboJackets/robocup-software/tree/justin/radio-test-nov18/firmware/common2015/drivers/cc1201/cfg/cc1101-compatible) (commit cb7adfe567cbbdece6a8c4208f54ae82aba4c478).

justbuchanan, 2015-11-19:
I've been messing around with this some more and I've noticed a few things:
* bumping the data rate from 250 -> 300 ksps with the previous config worked fine
* changing from 2gfsk to 4gfsk and dropping back down to 200ksps made the CRC check fail about 3/4 of the time
* I then enabled data whitening and the CRC fails on 1/3 packets now

justbuchanan, 2015-11-19:
I've got RX working now too - I can send packets from the cc debugger to a radio connected to the mbed.  This is the commit that has this working: 1b0db665be40bc97b7292bab9b23ef35e02d20f2.  The things I changed to get it working were:
* The test board had the radio interrupt line connected to the wrong MBED pin
* The CommTaskThread main method was waiting on a signal that is never sent by anyone
* I uncommented a few lines so that CommTaskThread repeatedly sends the `SRX` strobe to put the radio in RX mode.

jjones646, 2015-11-21:
@justbuchanan Sweet! Sounds live you've made quite a bit of progress.

The `CC1201.freq()` issue might have something to do with `freq_offset` being a `uint16_t` that gets bit shifted down by 18 bits...The math within the function itself looks like it lines up fine with what's in the CC1201's user guide, but I guess you'd have to multiply `freq_offset` by 40E6 before dividing by 2^18 in order for those bits to be kept. The function itself returns frequency/1E6 as it stands now.

justbuchanan, 2015-11-25:
I'm closing this issue now that packet RX and TX is working (somewhat).  I'll continue working on improving the config and try to fix the RX<-->TX state transition issues we're having.

closed 2015-11-25 by justbuchanan