RoboJackets/robocup-software #491: stream-based logging for firmware

justbuchanan, 2015-11-04:
The more or less "standard" way of doing logging in C++ is with streams.  We implemented something this a while ago, but I think some of it got lost due to merge conflicts.  It'd be nice to be able to something like this in firmware:
~~~{.cpp}
LOG(WARN) << "Something happened to: " << myObject << endl;
~~~

The real power here is that you can implement `operator<<(ostream)` on different classes and very easily print them to the log.

justbuchanan, 2015-11-12:
As discussed yesterday, the version of the firmware logger on the `master` branch has this feature, but it looks like it got lost on the `robot2015-firmware` branch.  We should just be able to copy the version on `master` to `robot2015-firmware` and verify that it works.

This will need to be tested on the MBED since we can't run our firmware code on the computer itself.  @hmistry1, we can get you an MBED to use at the next meeting if you remind me.

justbuchanan, 2015-11-18:
Are you having trouble getting the branch onto GitHub or making a pull request once the branch is there?

And sure, we can get you an MBED to test on next time.

JNeiger, 2016-02-17:
So I overloaded the `log(...)` function within the [`logger.hpp`](https://github.com/RoboJackets/robocup-software/blob/robot2015-firmware/firmware/common2015/utils/logger/logger.hpp) class to allow for a `std::ostream&` type overload (mirroring [this](https://github.com/RoboJackets/robocup-software/blob/master/firmware/common2015/utils/Logger/logger.hpp#L40) function in the old code).

Since both of these are going to be called through the [`LOG(...)`](https://github.com/RoboJackets/robocup-software/blob/robot2015-firmware/firmware/common2015/utils/logger/logger.hpp#L24-L25) macro, I think the next step is to "overload" the macro so it can call both depending on there are any arguments besides the log level. Most likely loosely mirroring this style of [design pattern](http://stackoverflow.com/questions/8886812/overload-c-macros). 

@justbuchanan, Would that be the right way to go on this or is there a better way?

justbuchanan, 2016-02-17:
Yeah that sounds about right.  We could just name the stream-based macro function to something slightly different.

One thing I just thought of that may cause problems for stream-based logging is thread-safety.  You'll notice in our current printf-style log function, we use a mutex to keep the log from being written to by multiple threads simultaneously and making a mess of the messages.  I'm not sure offhand how we could implement this with stream-based logging since the log() function returns a stream and the actual writing to the stream happens outside of the log() function.  There may be a workaround, but I'm drawing a blank at the moment.  Any thoughts?

@barulicm, any ideas for threa-dsafe stream-based logging?

JNeiger, 2016-02-17:
I was looking around and we may have to throw some sort of `flush()` function after the writes, it will solve the problem of collisions within the ostream output.

A temporary ostream will have to be created on each instance of a `LOG(...)` call. The insertion operator will be overridden to instead push the ostream input into a string buffer. This will allow for the entire string to be saved when chaining multiple `<<` operations together in a single line. Once the log line has been executed, the temporary ostream will then go out of scope. Within it's deconstructor, it can place the fully formed buffer onto an underlying global log queue object. A `flush()` call can then be called after each write. The flush can contain the mutex lock and only allow a single line written at once. 

This could also be used to time the other `printf` style log function since those could collide too with our stream-based logger, if I understand the code right.

I'm sure there's some magic we could do with the `LOG(...)` define and a `ENDLOG' (or something) define to wrap this up into a simple to use format.

justbuchanan, 2016-02-17:
I like it!  I think that will work perfectly for us.  I was looking through the [google logging library](https://github.com/google/glog) and they use the same trick of having the macro expand to a temporary ostream object that writes its contents when it is destructed.  I made a quick gist to try this out: https://gist.github.com/justbuchanan/7eb692b3e7ba946b2698.

Printf, by default, writes the string it's passed to a temporary buffer rather than straight to stdout for performance reasons.  It only automatically writes the temporary buffer to stdout when it reaches a newline.  The `fflush()` function just ensures that the string that was just printf()'d actually gets written to stdout, regardless of whether or not it ended in a newline.

Right now the `log()` function is threadsafe because it uses mutexes, but there are some calls to printf() (mostly in commands.cpp) that can happen in parallel with calls to `log()`, which causes some garbled output.  One solution to this would be to make our own threadsafe version of printf something like this, and never use printf or cout anywhere else.  Our log functions and printouts in commands.cpp would then all use the threadsafe printf.

~~~{.cpp}
void printf_ts(const* fmt, ...) {
    static Mutex m;
    m.lock();
    printf(fmt, __VA__ARGS__);  // note: use of __VA_ARGS__ like this isn't right, see logger.cpp
    fflush();
    m.unlock();
}
~~~

JNeiger, 2016-02-24:
I created a little [test program](https://gist.github.com/JNeiger/c49a703158922277b3bf) to make sure the solution works, but I can't seem to cause two `printf` calls to overlap like their `cout` counterparts. 

Is this just because of my implementation within windows? Or do we not need to lock the `printf` calls like we thought?

I've been looking deeper into printf and it seems that ([At least in C and the pthread implementation](http://linux.die.net/man/7/pthreads)) `printf`, and subsequently `write` are thread-safe. I wasn't able to find any documentation this detailed on C++'s version of `printf` but I'm assuming they follow the same POSIX standards.

justbuchanan, 2016-02-24:
I've done some looking around and like you said, `printf()` is thread-safe in POSIX.  From what I can tell, it's also thread-safe on Windows and on the MBED, which I wasn't aware of.  I think part of the reason why we've seen some weird logging is because there are two calls to `printf()` in our `log()` function, which gives the opportunity for another thread to call `printf()` in-between.  There's also something strange that happens when the `Console` class first starts up, but that's an unrelated issue.

I'm pretty sure `printf()` in C and C++ are exactly the same.  When you call it in C++, you're calling the function provided in the exact same libc library as if you were writing C code.

It looks like we can probably do everything without mutexes since `printf()` has them builtin.  To fix our current issue, one option is to merge it into one call to `printf()` instead of the 2 we currently have in `log()`.  To do that, we might be able to do something like in [this stackoverflow answer](http://stackoverflow.com/questions/7228873/how-can-i-add-a-new-argument-to-the-existing-variable-argument-list).

closed 2016-03-10 by justbuchanan