RoboJackets/robocup-software #493: Create basic sounds for the robot's speaker

jjones646, 2015-11-04:


jgkamat, 2015-11-05:
@joshhting

ra4king, 2016-02-10:
@jjones646 Is writing audio data using an interrupt-based mechanism the most efficient way of playing audio? A 44.1 KHz audio source would have to have its interrupt fired 44,100 times a second!

jjones646, 2016-04-04:
Well, it all comes down to how much we value audio - and the tradeoff of quality with system resources. The main issue is that it has to play nicely with the rtos setup, so I'm not sure if other approaches would work out.

justbuchanan, 2016-05-18:
@ra4king how far were you able to get on this?  And do you think you'll be able to get it finished somewhat soon or should we try and find someone else to work on it?

jgkamat, 2016-08-29:
Closed in favor of rj/robocup-firmware's issue, linked above.

closed 2016-08-29 by jgkamat