RoboJackets/robocup-software #503: Add a short contributing cheatsheet to complement our existing contributing guide page

justbuchanan, 2015-11-12:
Here's what I have from yesterday's meeting that will go on it:

~~~{.md}

* Choose a project
  * if there's no a GitHub issue for it yet, make one

* Make a new branch
  * don't work on the master branch
  * `git checkout -b <name>/<topic>`

* Research the topic
  * Use Google, Stack Overflow, RoboCup documentation site
  * Ask questions
    * comment on a GitHub issue
    * gitter chatroom

* Code!

* Cleanup
  * run code formatter: `make pretty`
  * Comment anything confusing
  * Simplify code as much as possible

* Submit a Pull request
  * Find your branch on GitHub, then hit the pull request button
  * give it a descriptive title
  * mention what GitHub issues are closes (if any) - use a # sign and GitHub will make it a link

~~~

jgkamat, 2015-11-12:
maybe we could have a flowchart?

[image]

justbuchanan, 2015-11-12:
I like it!  Want to take this over and add it somewhere on our docs site?  Just curious, what software did you use to make this?

A couple suggestions:
* it might be good to add a block before "Start Working!" that mentions brainstorming a solution and asking for feedback before doing too much coding
* Also might help to style the shell commands differently from the rest of the text just to make it a little clearer

jgkamat, 2015-11-12:
Oh, this was actually just Violet UML creator :stuck_out_tongue: 

I'll make this way better, this is a pretty rough first draft...

justbuchanan, 2015-11-12:
Cool, sounds good!

jgkamat, 2015-11-26:
This is what I have now, I converted it to LibreOffice Draw to try to get more formatting, Still a WIP though

[image]

justbuchanan, 2015-11-26:
Nice, looks pretty good to me!

jgkamat, 2016-12-11:
This eventually became this: https://robojackets.github.io/robocup-training/slides/8#/sec-title-slide

And the docs page: https://robojackets.github.io/robocup-software/t20168.html

closed 2016-12-11 by jgkamat