RoboJackets/robocup-software #508: Add legit radio test-rx

justbuchanan, 2015-11-19:
Right now it's a loopback test where a packet is placed in the rx buffer, then read back again.  This is a good test that the radio interface works, but doesn't actually test rx.

cc @jjones646

jjones646, 2015-12-19:
@justbuchanan I added a `loopback` test to the `radio` command a while back. Increase the log level a bit and you can see it doing the full ack sequence. So `test-rx` can be changed to an actual air packet now without losing this.

`radio loopback <message>`

jjones646, 2016-02-04:
@justbuchanan Are the new commands stable enough so this can be closed?

justbuchanan, 2016-02-04:
Yeah I think so.  I'll merge my PR that changes to the DISCOVER port into the firmware branch.

closed 2016-02-04 by justbuchanan