RoboJackets/robocup-software #511: Running `make robot2015-prog` while screened into a robot corrupts output

justbuchanan, 2015-12-11:
Not sure what the cause is, although it might be because we switched to a higher baud rate (57600) for the serial port.  @jjones646 any ideas?

This is an issue on the current robot2015-firmware code

jjones646, 2015-12-19:
This has bugged me for a long time...I know exactly what you're talking about, and it's happened to me in many different forms with different configs/baudrates. The mbed's USB serial connection is controlled through the mbed interface chip on the back. So a serial character goes pc -> mbed interface -> lpc1768, and vice-versa.

The mbed interface chip is also how the lpc1768 is loaded with a new binary at startup. Some info is sent back to the pc from the mbed interface chip when a new binary is loaded onto the mcu. This is solely from the mbed interface's firmware. I believe this only happens when loading a new binary - not on a reset. I've played around with pyserial before and have actually read what it sends back. If I remember correctly, it's json formatted. Not entirely sure how I did this, but I know it can be done.

The only half-reasoned explanation I've come up with so far is that some of this is picked up by screen and treated as a control/special character/sequence. Changing to a 57600 baudrate could be causing some corruption here and the characters being received are some arbitrary character since they'd still be send at 9600? If there's a way to show every received/sent character with screen, that may be a good way to get some insight into what's going on so we can find out 1) if this is actually the cause for the corruption, 2) if anything else is going on that we don't know about, and 3) how to get around this while screened into a robot.

jjones646, 2015-12-20:
@justbuchanan I played around with some things today on this. The changes are in the `robot2015-firmware` branch. The main one being the udev rules file in `./util`. Basically added automounting for an mbed's usb flash memory. It's kinda shooting in the dark, but seems to work for me - the text hasn't gotten corrupted since the change. The reasoning behind all of it was to remove any mounts/unmounts in the mbed-copy script itself.

Give it a go and see if the udev rules work outside Ubuntu. It's not perfect and still fails to remove a directory every now and then under `/mnt/robot2015` if you disconnect an mbed while screened into it, but that's the only problem I've came across yet. `sudo` also shouldn't be needed anymore during `make robot2015-prog` or when doing `screen /dev/mbed0 57600`.

jjones646, 2016-01-24:
This is semi-fixed in the firmware branch. The usb virtual serial connection seems to depend quite a bit on the mbed itself. And updating the mbed interface firmware to the latest version on the mbed website is critical for this - it seems that the mbeds were not at this latest version when we initially received them.

For anyone paying really close attention, they would have noticed that I quickly reverted everything in my previous comment above. Long story short: the udev rules stopped block devices from loading during boot...so your computer wouldn't boot if you restarted it.

I've worked in a slightly different direction on this issue, however. I've gotten it to work perfectly before with multiple connected mbeds - that's it, `make robot2015-prog` while screened into the mbed and everything reset like it's supposed to with good console text. However, I can't get it to consistently do that between unplugging and plugging it back in.

justbuchanan, 2016-01-24:
@jjones646, I'm having a little bit of a different issue now that's related to this.  I'm consistently seeing output as shown in the screenshot below:

justbuchanan, 2016-01-24:
[image]

I did a `git bisect` and it says that commit 7844be0 is the culprit.  Can you take a look and see if you see something that might cause this?  Btw, I only see the huge chunk of question marks when a radio is connected.

jjones646, 2016-01-25:
This was happening the other day and has something to do with the fpga config failing because there's no bitfile on the mbed. I'll look into it.

jjones646, 2016-01-25:
@justbuchanan check out the stuff I just pushed. Should work for every possibility of startup results now.

closed 2016-02-03 by justbuchanan