RoboJackets/robocup-software #519: Checkstyle does different things on mac vs ubuntu

ashaw596, 2016-01-22:
For some reason checkstyle on mac and ubuntu are slightly different. For example, 85721a7 passed checktyle on mac, but on ubuntu running make pretty resulted in, af30f23 which fixed two files.

justbuchanan, 2016-01-22:
Yeah this is a known issue, I don't really know how to fix it though :/.  The reason this happens is that homebrew pulls a more up-to-date version of `clang-format` than ubuntu does and there are slight differences in the formatting they apply.

ashaw596, 2016-01-22:
oh meh. .... Man this lack of ubuntu support for up to date things are so annoying.

jgkamat, 2016-01-22:
Maybe clang-format has a backwards compatible mode? Newer versions shouldn't produce different output imo, unless there is a major version bump.

justbuchanan, 2016-01-22:
@jgkamat, not that I could find unfortunately :/.  Not really sure how to fix this since it's tough/impossible to run the same version of clang-format on each platform.

One option would be to have the ci setup automatically run the formatter and push a commit to fix it.

jgkamat, 2016-01-23:
I don't like the idea of pushing automatically, because that could cause conflicts (multiple pushes, etc) if commits are added quickly. I would like the idea of printing the diff in the logs, and a command to easily apply the diff, though. If you think thats good, I'll go ahead and figure out how to do that.

justbuchanan, 2016-01-24:
That sounds good to me.  We could have it run a checkstyle, then output a patch file to apply if anything fails.  This will potentially require some changes to `stylize`, although it might be difficult.  I'll look into it and see what I can figure out.

justbuchanan, 2016-01-28:
I opened an issue for this on the stylize repo [here](https://github.com/justbuchanan/stylize/issues/8).  The plan is to add an option to `stylize` so that it outputs a patch file (basically a git diff) so that fixing a broken style will be a one line command that downloads the patch file from circleci's build artifacts and applies it with `git apply`.

PR #533 (2016-02-01): CI server now generates a patch that can be applied to fix style
closes #519
closed 2016-02-01 by justbuchanan