RoboJackets/robocup-software #530: Add ability to save and restore robot locations to soccer

ashaw596, 2016-01-27:
Currently, we can drag robots and reset the robot positions. It would be great if there was a way to save the current state of the robots and restore it in to allow easier testing plays and things in simulator.

justbuchanan, 2016-01-27:
Duplicate of #363

ashaw596, 2016-01-27:
Its not the same. Saving the current state of the simulator to a config file is an issue of persisstance which is significantly more complicated. This is saving the location in soccer in memory and the restoring it using the currently already implemented communication protocol between soccer and simulator.

ashaw596, 2016-01-27:
Also the other issue #363 sorely lacks detail about the desired feature

PR #550 (2016-02-25): Quicksave and Quickload
Solves issue with Stop Robots not working. Adds quicksave and quickload functions under Simulation. Closes  #530
closed 2016-02-25 by justbuchanan