RoboJackets/robocup-software #537: Add support for building/loading firmware from Mac OS

justbuchanan, 2016-02-04:
I haven't looked into this much, but @jjones646 found some hacks to get things to build.

The gcc-arm compiler/tools suite can be installed with `brew cask install gcc-arm`.  Our CMake setup needs some changes for this to work though.

jjones646, 2016-02-04:
I was actually playing around with the cmake scripts earlier today. In an attempt to use as many cmake cross compiler things as possible. I also have a readme file that I started making about halfway through setting up my laptop. Wasn't really planning on pushing all of it yet, but I'll try cleaning a bit of it up and open a pull request so we can see how it checks out with circleci.

jjones646, 2016-02-07:
I opened a [pull request](https://github.com/RoboJackets/robocup-software/pull/538) for this. It's not ready, but it is very close.

closed 2016-03-22 by jjones646