RoboJackets/robocup-software #542: Implement ball placement for 2016 technical challenge

justbuchanan, 2016-02-16:
One of this year's technical challenges is to receive ball-placement commands from the referee system and have our robots carry them out.  The intention is that in the future, ref calls will be all automated and the robots on the field will set the ball up in the right position for restart plays.

Details here: http://wiki.robocup.org/wiki/Small_Size_League/RoboCup_2016/Technical_Challenges/Autonomous_Ball_Placement.

jgkamat, 2016-02-16:
Would this be implemented as a simple tactic? If so, it shouldn't be too hard to do (we might have to add a 'let go of the ball' skill).

justbuchanan, 2016-02-16:
Yeah more or less.  I don't think this would be too hard to get running.  We'd need to add support for the new messages from the referee system and add a new game state and a variable to hold the desired ball position.  Then we can add a new play that's enabled only in that game state and have it do the work.

justbuchanan, 2016-02-16:
Here is the relevant protobuf file in the referee system.  Note that it's not on the master branch yet.

https://github.com/RoboCup-SSL/ssl-refbox/blob/remote-control/rcon.proto

chachmu, 2016-04-17:
Is there a way to control the simulated refbox to send placement commands? Also I think they have moved the protobuf file to the master branch now.

PR #651 (2016-05-22): Implement Ball Placement Challenge
Closes #542.
closed 2016-05-22 by justbuchanan