RoboJackets/robocup-software #546: Calls to printf() in firmware aren't threadsafe

justbuchanan, 2016-02-22:
Some corruption is caused in the log output due to calls to `printf()`.  Probably the best solution would be to add a threadsafe wrapper around printf and call it from `log()` and any other place we want to print.

Somewhat related to issue #491 

cc @JNeiger, @jjones646

jjones646, 2016-02-29:
From issue #491, yes, `printf()` is already thread safe. Thread safe in the sense that two calls to `printf()` will not cause a memory race condition and crash the microcontroller. However, there is no guarantee about keeping buffers in order.

The mutex used in the log function ensures that two 'parallel' calls to `log(...)` will be sent to `printf()` at different times. The log function also ensures that a new line character is appended to the end and the entire thing is flushed before removing the mutex lock. This is because there is additional buffering on the mbed interface chip that will only flush the uart buffer when filled over a threshold or at a newline character.

For all firmware code I've written, I've tried very carefully to only make `printf()` calls at times when it's reasonably safe to say that the mbed interface's uart buffer is empty, and that no other threads have recently called `log(...)`. All `printf()` calls in command.cpp are there because the thread's priority is increased above all others when executing a line. This essentially suppresses all `log(...)` calls during that time - and it's safe to assume that everything will be executed in order for any of those functions.

The only real solution to all of this is to make our own `printf()` function that is aware of the rtos threads. As explained [here](https://github.com/RoboJackets/robocup-software/issues/491#issuecomment-185375853), except you don't want to wrap a mutex around `fflush()` because that can still happen if the stdout buffer is filled beyond a certain threshold (don't take that as fact, but I'm pretty sure). You'll have to wrap the mutex around everything - pretty much exactly how the `log()` function is written.

Lastly, a lot of good stuff can be found in [external/mbed_lib_build_tools/libraries/mbed/common/retarget.cpp](https://github.com/mbedmicro/mbed/blob/master/libraries/mbed/common/retarget.cpp), so it may be possible to redirect stdout to our own defined buffers where everything could be handled correctly.

JNeiger, 2016-03-07:
I ended up making a general [`log_printf` function](https://gist.github.com/JNeiger/555a20c2418197121f68) that I think I will help solve both this and #491. I think I have everything in there setup right, but I just want to make sure.

justbuchanan, 2016-03-07:
Looks good to me.  Rather than make this its own function, you could probably just change the implementation of our current `log()` function to be more like this.  Using a temporary buffer so that we only have to write to stdout one time per log should fix our issues.

closed 2016-03-17 by justbuchanan