RoboJackets/robocup-software #552: mbed OOM

justbuchanan, 2016-02-27:
I've tracked the firmware segfault down to a lack of available RAM.  When I startup the MBED and type a command at the console, it usually will immediately segfault and reboot.  It happens at the `args.reserve()` line [here in commands.cpp](https://github.com/RoboJackets/robocup-software/blob/master/firmware/robot2015/src-ctrl/modules/commands/commands.cpp#L992).  GDB doesn't show a valid stack trace, so I think it's making a heap allocation, which causes the heap to collide with the stack and everything goes to shit.

I've also noticed that if left alone for long enough, it will segfault by itself and reboot.  I can't confirm what line of code this happens on, but I assume it's a similar issue with lack of memory.

I'm hoping that there's a large memory leak somewhere that we can fix and have everything work great.  If that's not the case though and we're actually using all of the RAM, we're going to have to scale back a good bit.

Any thoughts, @jjones646?

justbuchanan, 2016-02-27:
The [MBED Memory Model](https://developer.mbed.org/handbook/Memory-Model) documentation says that if you allocate too much memory on the heap, this will happen.  See the "Collisions" section.

jjones646, 2016-02-27:
Each thread has it's own stack. So that makes things exponentially more complex to track down.

GDB may not be able to provide a valid stack trace because of it. I don't know enough about the inter-workings of the RTOS framework to say that for sure, but it's definitely something to consider here. That line is called very early after a signal would be released for that thread - and if you manually follow up the code for what a stack trace should look like, it turns out to be just `execute_line()`. GDB may show some weird things for all of this since each thread's memory space is from a heap allocation when first started.

They do have some rather rudementary ways to check for this kind of thing. You can `#define OS_STKCHECK`, and a stack overflow forces an endless loop in the `os_error()` function. You can read all about it and some other things [here](https://www.keil.com/pack/doc/CMSIS/RTX/html/_thread_config.html#StackCheck). At that point, you'd know for certain whether it was a thread stack overflow or truly a heap allocation of all available RAM failing.

I've looked into a few ways to run memory checks on the firmware. Apparently there are ways. I tried it once for like 5 minutes one day, but wasn't very motivated at the time. However, it is something we should probably look into more seriously.

jjones646, 2016-02-28:
So it looks like `#define OS_STKINIT` may be the thing to use, it says `#define OS_STKCHECK` is for checking stack overflows at a thread switch.

justbuchanan, 2016-02-28:
Good point, I completely forgot about each thread's stack when I was accounting for memory usage.  I think if we rearrange a bit, we can cut out the `CommTaskThread` since it really just initializes the `CommModule`.  Doing that will save us a stack, which may be enough.

I'll look into the stack check macros.  I tried to use the `-fstack-protector` flag for `g++`, but it didn't seem to be supported for MBED.

justbuchanan, 2016-02-28:
I tried changing up CommTaskThread so that it initializes Comm stuff, but is run on the main thread instead of its own thread.  I haven't had it segfault since then.

I'll clean it up a bit and send you a PR.

jjones646, 2016-02-29:
I don't believe `-fstack-protector` is of much use at a kernel level, and would make the stacks bigger if used.

I image there some weird linker things happening. The threads are defined with macros at the lowest level, so there may be some quirky things going on there. Since the the common2015 library contains threads and is compiled at a different time than that of main, the preprocessor may treat those slightly differently. So I think your idea of removing the CommTaskThread is def a good one. Especially when considering that all of those operations are handled in the library threads.

We can also change each thread's stack allocation. So any threads that don't need a big stack can be reduced. Since the [Mail queues](https://developer.mbed.org/handbook/RTOS#mail) are heap allocations, this includes almost every thread. With the except of whatever is happening when `Console::Init()` is called. That's an area that could be looked at because I have no idea what actually happens with the whole `shared_ptr` ordeal. But I do know 800+ bytes are allocated somewhere when the object is created.

commit dee58a93 (2016-03-01): Comm init happens on main thread, not a separate one.  closes #552

By reducing our number of threads by one, we saved one whole thread stack that had to be allocated.  This reduced our memory usage by enough that we dont crash due to lack of memory anymore.
commit 52365d8d (2016-03-01): Comm init happens on main thread, not a separate one.  closes #552

By reducing our number of threads by one, we saved one whole thread stack that had to be allocated.  This reduced our memory usage by enough that we dont crash due to lack of memory anymore.
commit c92a29ef (2016-03-14): Merge pull request #558 from RoboJackets/justin/comm-task-main-thread

Comm init happens on main thread, not a separate one.  closes #552
closed 2016-03-16 by justbuchanan