RoboJackets/robocup-software #554: to-Python converter throwing errors on make run-sim

jcarn, 2016-02-28:
The following errors are thrown immediately before the soccer window pops up and runs normally:

Initializing embedded python interpreter...
_frozen_importlib:321: RuntimeWarning: to-Python converter for OurRobot* already registered; second conversion method ignored.
_frozen_importlib:321: RuntimeWarning: to-Python converter for OpponentRobot* already registered; second conversion method ignored.
_frozen_importlib:321: RuntimeWarning: to-Python converter for SystemState* already registered; second conversion method ignored.
No logs/ directory - not writing log file

jgkamat, 2016-02-28:
removing the `register_ptr_to_python` statements in `robocup-py.cpp` seems to solve the problem (and they don't seem to affect usage). Maybe a newer version of boost removes the need for those statements?

jgkamat, 2017-01-13:
Aah, I duplicated this in #828. Its fixed now =)

closed 2017-01-13 by jgkamat