RoboJackets/robocup-software #591: Create a UI popup when Python Errors Occur

ashaw596, 2016-03-29:
It would be nice to have some type of alert so we're not confused when the robots just don't move.
We'd have to figure out how exactly to do this.

jgkamat, 2017-01-13:
The hardest part of this issue is figuring out how to intercept the python errors properly. Once that is done, we can either:

1. Use python to print the message directly (bad)
2. Pass it up to C++, so it can be displayed in a log somewhere (nice)

Ideally, we wouldn't have a popup, so repeated errors won't cause a popup cascade. I'm willing to take feedback on that though.

chachmu, 2017-01-13:
Maybe have an generic "python error" message for the status indicator at the top so you can see it from within soccer but it doesn't get in the way.

jgkamat, 2017-11-29:
Closing in favor of #1034, which is a duplicate for this.

closed 2017-11-29 by jgkamat