RoboJackets/robocup-software #592: Update radio config to make use of cc1201's speed and features

justbuchanan, 2016-03-31:
Right now we're using the same config on the cc1201 as we used before on the cc1201.  We should be able to change some parameters to get things a good bit faster.  We also may want to check out things like data whitening, etc.

jjones646, 2016-04-04:
The only real reason for moving to the CC1201 was for being able to use 4-GFSK modulation. That's where the most speedup would happen, but it also means no compatibility with the current base station. The physical layout of the PCB also become more important for signal quality with 4-GFSK.

justbuchanan, 2016-04-04:
Sounds good.  I've got a config from testing a while ago that uses 4-GFSK and seemed to work pretty well.

As for compatibility with the current protocol, I haven't been able to get the CRC to pass on a packet sent from the old base station to a new cc1201 board.  Not sure what the deal is with that because I read through the datasheets thoroughly and had it configured appropriately from what I could tell.

guyfleeman, 2018-06-04:
moved to decawave, closing

closed 2018-06-04 by guyfleeman