RoboJackets/robocup-software #593: Fix build warnings in build related to boost python

justbuchanan, 2016-04-01:
I've messed around with this, but haven't had any luck fixing it yet.  It's incredibly obnoxious though to see tons of these scroll by on each build:

~~~{.sh}
In file included from /usr/include/boost/type_traits/ice.hpp:17:0,
                 from /usr/include/boost/python/detail/def_helper.hpp:9,
                 from /usr/include/boost/python/class.hpp:29,
                 from /usr/include/boost/python.hpp:18,
                 from ../soccer/gameplay/GameplayModule.hpp:6,
                 from ../soccer/main.cpp:2:
/usr/include/boost/type_traits/detail/ice_not.hpp:17:71: note: #pragma message: NOTE: Use of this header (ice_not.hpp) is deprecated
 # pragma message("NOTE: Use of this header (ice_not.hpp) is deprecated")
~~~

jpfeltracco, 2016-08-30:
I've looked at this too and tried setting the boost include directory as a system directory, documentation [here](https://cmake.org/cmake/help/v3.0/command/include_directories.html) so that warnings would hopefully get ignored but it didn't work. I messed around with some flags to suppress deprecation warnings as well to no avail.

A bit of searching has come up with many complaints on other people's projects about similar bugs in CMake 1.60. I believe our particular warnings have been fixed in 1.61 as per [this bug report](https://svn.boost.org/trac/boost/ticket/12029), but Arch doesn't have 1.61 yet (let alone Ubuntu).

We should probably just close this and live with the warnings because it doesn't appear like something we can fix until boost gets updated.

Thoughts @jgkamat?

jgkamat, 2016-08-30:
So this is kind of an interesting issue because I can repoduce it on arch, but I have never been able to reproduce it on debian. ([debian is on boost 1.61 btw](https://packages.debian.org/stretch/libboost-all-dev), but I didn't see this previously either). I want to keep this open until we can confirm that 1.61 indeed fixes it for arch users (I hope it will, it should get promoted soon). 

Even if we could do something about it, waiting for 1.61 is probably the cleanest solution.

jpfeltracco, 2016-08-30:
I just manually installed 1.61 from staging in Arch and I can confirm the boost warnings are gone.

I'm going to go ahead and close this, we can reopen if it crops back up.

closed 2016-08-30 by jpfeltracco

jgkamat, 2016-08-30:
Cool, glad to know its finally fixed! :smile: