RoboJackets/robocup-software #603: Scoring while running a play causes PathPlanning failed errors

jcarn, 2016-04-06:
This isn't urgent at all, but it is a bit of an annoyance. When an offensive play is ran and the robots score, they start throwing PathPlanning Failed errors. They continue doing this even if you remove the ball from the goal, and often freeze in place. After some random amount of time they can unfreeze themselves and start acting normally. Disabling then re-enabling the play seems to fix the error. Once again, this doesn't seem to affect actual gameplay at all

justbuchanan, 2016-04-06:
@ashaw596 any ideas?

`git bisect` might be a good tool to use to track this down, assuming it was introduced semi-recently.

ashaw596, 2016-04-06:
I'll take a look. I was pretty sure it was mostly just the ball being in an unreachable location, but I'll check.

ashaw596, 2016-04-06:
The reason its popping up now is just that I added that debug statement to the pathplanner, but I'll check the freeze in place bug.

justbuchanan, 2016-06-22:
@ashaw596 any progress on this?

ashaw596, 2016-06-23:
I mean it's a basic problem with our path planning system not being able to
plan into obstacles. I'm not sure what to do...

justbuchanan, 2016-06-23:
Right, I don't think that part's the issue.  It's fully expected that our robots get confused when they try to go for the ball when it's in their goal zone.  I think @jcarn had a problem where they were still confused after he removed the ball from the goal zone and placed it in a valid location.  Is that still an issue?

ashaw596, 2016-06-23:
Oh, I'll try to reproduce on my system, but I feel like the freezing bug was mostly fixed earlier in. https://github.com/RoboJackets/robocup-software/pull/673.

jgkamat, 2016-10-02:
Is this referring to the `RRTPlanning Failed` messages in the terminal? I seem to be getting these pretty often...

jgkamat, 2017-01-22:
@jcarn are you still seeing this issue? I think this was fixed in the path planning revamp that happened last semester.

jcarn, 2017-01-24:
It looks to have disappeared.

closed 2017-01-24 by jcarn