RoboJackets/robocup-software #623: FPGA.send_config() has potential infinite loop

justbuchanan, 2016-04-12:
If something's broken, the `do {} while ()` loop in this method might never terminate.  We should have it time out after a set number of iterations and return an error.

jjones646, 2016-04-13:
Did this happen to you? This seems like something that requires the perfect storm to happen.

`FPGA::send_config()` really needs to be private - it was never really meant to be called outside the class..

`FPGA::configure()` *should* return before calling `FPGA::send_config()` if there's something wrong with the minimal required hardware. If you send a file that doesn't have `0x00` in it, that could possibly cause a stall - but `initB` should break out of the loop for that case.

jjones646, 2016-04-13:
@justbuchanan When you get some time, send `MBED.HTM` as the config file. If it doesn't stall when doing that, then this would be solved by just making `FPGA::send_config()` private - or just putting it inside `FPGA::configure()`. Just did a recursive search and `FPGA::configure()` is the only place it's used - and probably should be used from considering calling it the wrong way means no motors.

justbuchanan, 2016-04-13:
It didn't actually happen to me, I think I'm just paranoid haha.  I had a similar loop in CC1201 a while ago that waited for an IO to change that in some cases wouldn't happen - it took a while to track down exactly where the issue was.  I took a look at the code again and you're right, it'll always end because the loop stops when it hits the end of the file.

jjones646, 2016-04-18:
@justbuchanan  So we switched out the FPGA today in favor of one that has more logic gates. Guess what happens if you try sending it a bitstream file that was synthesized for our current FPGA part number instead of it...this.

It does eventually time out, but it's 30+ seconds at least. It's an easy fix in the code actually, and I just did it. But none of that is pushed to here yet.

justbuchanan, 2016-04-18:
Hahaha that's awesome.  Good to know I wasn't totally off-base here :).  Should be an easy enough fix

jjones646, 2016-04-18:
Fixed

jjones646, 2016-04-18:
jk...see [here](https://github.com/RoboJackets/robocup-software/commit/398bc9b3b3feeee76a6a01383a4f9eb41bce1240#commitcomment-17136667)

jjones646, 2016-04-18:
Here's a quick update of where the fpga config starts failing. It's an issue at the moment, but wouldn't consider it a bug worth looking into just yet. I won't go into too many details, but it's almost certainly the SoftwareSPI to blame here - and that's getting changed once the new boards are in.

bad -> 165a6c5 - revert main.cpp changes from avr branch merge
can't compile -> 1cc30b2 - Merge branch 'justin/avr-fixes'
good -> 2444d0f - Merge branch 'justin/radio-feb29'

jjones646, 2016-04-18:
Scratch that, see #653

PR #654 (2016-04-18): FPGA configuration fixes and updated error LED states
These fixes have been confirmed in hardware. Renamed a few things to make it more clear as to what's going on in `FPGA::configure()`. Also rolled out the proper error led states for all of it now that the io expander is able to be tested on.

I've also tested the ls command with the control board connected or not connected with all of this. So, that's apparently fixed now I believe. Tested 3 different mbeds with a combination of files on them, and it all checks out.

fixes #653 
fixes #623 
fixes #639
closed 2016-04-18 by jjones646