RoboJackets/robocup-software #633: Implement "self test" functionality for firmware

justbuchanan, 2016-04-13:
The control board has a "self test" button on it that is intended to perform hardware checks and report if there are any errors.  It currently isn't setup to do anything, but we should write some code to make it functional.

@jjones646 any thoughts on exactly what this button should do?  We already have the red error leds set to light up for certain things, but a lot of those checks happen right now at startup and don't get checked again.  Maybe the button can just run the check again and update the leds for any new errors that have happened since startup?

jjones646, 2016-04-13:
Something that could be pretty useful would be to spin each motor ~1/4 turn in series - one at a time.

This also brings up what the dip switches will be used for too.

jgkamat, 2016-08-29:
Closed in favor of rj/robocup-firmware's issue, linked above.

closed 2016-08-29 by jgkamat