RoboJackets/robocup-software #637: Thread ids printed in the console don't match the "ps" command

justbuchanan, 2016-04-13:
The ps command shows thread task ids (generally single digit numbers), while the thread ids printed in several places in our code show up as huge numbers.  It looks like this is because the thread id printed in many places is from `Thread::gettid()`, which returns a pointer to `os_thread_cb` rather than the actual task id.

jjones646, 2016-04-13:
Any suggestions on steps to take for this? I guess we'll want to pick one or the other for both places.

IMO the single digits are definitely more intuitive, but I don't really know how to get that number using the mbed static methods.

JNeiger, 2016-04-13:
From what I can tell, `Thread::gettid()` returns a [P_TCB object](https://developer.mbed.org/users/mbed_official/code/mbed-rtos/file/bdd541595fc5/rtx/TARGET_CORTEX_M/rt_CMSIS.c) (Line 576). If we force cast the pointer to P_TCB then just grab the task_id from that, we can pull the single digit ones. 

It's almost the exact same thing we are doing [here](https://github.com/RoboJackets/robocup-software/blob/master/firmware/robot2015/src-ctrl/modules/commands/commands.cpp#L833).

```
P_TCB p = (P_TCB)os_active_TCB[i];

if (p != nullptr) {
       printf("%-4u\t%-5u\t%-5u\t%-6u\t\t%-10u\t%-10u\t%-10u\r\n",
                p->task_id, p->prio, p->state, p->delta_time,
                ThreadMaxStackUsed(p), p->priv_stack,
                ThreadNowStackUsed(p));

       num_threads++;
}
```

PR #646 (2016-04-14): ThreadID printouts use single digit numbers
Closes #637

I think I found all of them
closed 2016-04-14 by justbuchanan