RoboJackets/robocup-software #652: Stop play no longer works

jgkamat, 2016-04-18:
This is pretty important, we need the stop play to complete. Press the stop button in the top right, and assign a goalie.

Introduced in a10654f07fdf73f2951ae9a0081b865a23102880

```
ERROR:root:Play 'Stopped' encountered an exception: <class 'AttributeError'>.  Dropping and temp. blacklisting current play...
Traceback (most recent call last):
  File "/home/jay/Code/robocup-software/soccer/gameplay/composite_behavior.py", line 81, in spin
    bhvr.spin()
  File "/home/jay/Code/robocup-software/soccer/gameplay/composite_behavior.py", line 63, in spin
    super().spin()
  File "/home/jay/Code/robocup-software/soccer/gameplay/fsm.py", line 81, in spin
    StateMachine.spin(self)
  File "/home/jay/Code/robocup-software/soccer/gameplay/fsm.py", line 76, in spin
    self.transition(next_states[0])
  File "/home/jay/Code/robocup-software/soccer/gameplay/fsm.py", line 121, in transition
    state_method()
  File "/home/jay/Code/robocup-software/soccer/gameplay/plays/stopped.py", line 56, in on_enter_normal
    idle = tactics.stopped.circle_near_ball.CircleNearBall()
  File "/home/jay/Code/robocup-software/soccer/gameplay/tactics/stopped/circle_near_ball.py", line 32, in __init__
    for pt in self.get_circle_points(6):
  File "/home/jay/Code/robocup-software/soccer/gameplay/tactics/stopped/circle_near_ball.py", line 47, in get_circle_points
    for i in constants.Field.FieldBorders:
AttributeError: 'Field_Dimensions' object has no attribute 'FieldBorders'
```

Seems like something introduced in the move from constants.py to the Field_Dimensions object.

I don't think it would be too hard to fix, but I don't have time to look into it right now.

jcarn, 2016-04-18:
I'll check it out, talk to you tomorrow at the meeting if I have any questions

jgkamat, 2016-04-18:
Cool, thanks! Take a look at #566, that migration seems to be causing it. 

This is the field it was using before
https://github.com/RoboJackets/robocup-software/pull/566/files#diff-e66f525f29c7b1c22d55c141f04a8016L77

I'm not sure if there is a new alternative (you might have to make a new field in Field_Dimensions with the same contents as the old one).

chachmu, 2016-04-19:
That means its probably my fault, I will take a look tomorrow

PR #656 (2016-04-27): Fixes stop play bug
Added the missing FieldBorders array of lines that form the border to the FieldDimensions class and made the C++ to python converter reflect that change. Closes #652
closed 2016-04-27 by justbuchanan