RoboJackets/robocup-software #669: Path-planning fails near goal obstacle

justbuchanan, 2016-05-18:
I'm not sure exactly what the cause is, but I'm seeing some path-planning issues with the defenders near the goal zone.  Below is a screenshot showing the issue.  Robot 3 is supposed to be moving into position to block a portion of the red ball-projection triangle, but it isn't moving at all and has been there for several seconds.  I'm guessing it has something to do with the goal zone obstacle.  You can see in the "Tree" panel that robot 3's command velocity is essentially zero.

[image]

This is present on the master branch (I ran this on f4b82bebe5).

cc @ashaw596

justbuchanan, 2016-05-18:
It actually may not be related to the goal zone, I'm seeing the same problem in the screenshot below.  I ran soccer, set goalie to robot 2, then ran the TestDefense play.  Robot 5 is supposed to get into defense position, but is "stuck" where it started out.  It did finally move to where it was supposed to after several seconds, but it took a while.

[image]

jgkamat, 2016-05-18:
@ashaw596 This is the issue that I was showing you on the last day

You can reproduce this reliably by running the stopped play and moving the ball around a bit. Its usually a single robot (5 in my case) that gets stuck. Not sure why though.

ashaw596, 2016-05-19:
Hmm. Cool, If its reproducable I should be able to figure out whats going on. I'll take a look.

commit 9e1d3b50 (2016-05-19): Merge pull request #673 from RoboJackets/albert/fixRRTBug

Fix bug and switch to simplest acceleration algorithm fix #669
closed 2016-05-19 by justbuchanan