RoboJackets/robocup-software #671: Firmware "screensaver" stops working after a few minutes

justbuchanan, 2016-05-18:
The "screensaver"/imAlive lights on the MBED builtin LEDs work for a little while, then stop.  This may be related to PR https://github.com/RoboJackets/robocup-software/pull/588.

jjones646, 2016-05-25:
I vote to reverse the "screensaver" back to something simple. Something that doesn't need the timing requirements that is used now.

justbuchanan, 2016-05-28:
Sounds good.  How about just blinking one of the mbed's on-board LEDs?

justbuchanan, 2016-06-16:
I changed the strobing effect to a simple flasher on mbed's LED1.  It still stops after a minute or two and I can't find out why... The timeout is set to `osWaitForever`.

jgkamat, 2016-08-29:
Closed in favor of rj/robocup-firmware's issue, linked above (assuming this wasn't fixed at comp, if it was, feel free to close the linked issue).

closed 2016-08-29 by jgkamat