RoboJackets/robocup-software #672: CC1201 fails if other radios are transmitting when firmware boots up

justbuchanan, 2016-05-19:
Maybe we need to flush buffers and strobe it into RX after everything initializes

cc @jjones646

jjones646, 2016-05-19:
there's high speculation that it's due to the mbed missing the 1st edge triggered interrupt from the cc1201, but that hasn't been looked into yet.

justbuchanan, 2016-05-19:
Relevant google doc: https://docs.google.com/document/d/1oop82J2OscGVAEYbQcU4IoGHz5mU44n06LuA0Ib1iaU/edit

justbuchanan, 2016-05-23:
@jjones646 as of 293404dbee39, I think I have this all working!  You can see the commits for the full changes, but the highlights (that I remember) are:

* In `getData()`, I added an RX fifo flush if the radio says that no data is available
* In `CommLink.rxThread()`, the interrupt in is now registered *before* it waits for the call to ready().
* interrupt line is falling-edge triggered.  We've tried it both ways several times, but I've switched it to falling-edge and remove the `Thread::wait()` in `getData()` - everything seems to work
* removed all `SAFC` strobes.  On the transmitting side, I realized that it was sending this strobe whenever a *transmit* triggered an interrupt - I think this was causing it to calibrate according to bad data.  Removing this fixed the issue where the transmitter could only send ~5 packets before going out of calibration.  We may want to add calibration back in for valid RXs, but it's working fine without them for now.

justbuchanan, 2016-05-23:
Success!  The robot shows up and stays in soccer.

[image]

jjones646, 2016-05-24:
:clap:

> interrupt line is falling-edge triggered. We've tried it both ways several times, but I've switched it to falling-edge and remove the Thread::wait() in getData() - everything seems to work

Awesome! :+1: This is with the interrupt pin being configured with `0x06`?

> removed all SAFC strobes. On the transmitting side, I realized that it was sending this strobe whenever a transmit triggered an interrupt - I think this was causing it to calibrate according to bad [...]

I can't believe we overlooked something like this for so long. Now that you've pointed it out, everything is starting to make some kind of sense!

justbuchanan, 2016-05-24:
Yep, it's at `0x06`, it may need to change though (#677).

Yeah in hindsight it should have been obvious...  Glad it's finally fixed though, we should be able to move onto more fun stuff now!

commit 9857bec6 (2016-05-19): fixed #672
commit 5855fe26 (2016-05-20): Revert "fixed #672"

This reverts commit 9857bec6c068784c517cdd1c790a49b60fdf3f96.
PR #678 (2016-05-27): Lots of radio fixes
This fixes many issues we've had with our radios and enables robots to show up in soccer.  Fixes #672, #645, #644.

TODO:

- [ ] remove rate-limiting in USBRadio - we're currently only sending every other packet because the system can't seem to handle 60Hz. (#683)
- [x] Fix TODOs in USBRadio
- [x] USBRadio should address the correct robot - it's currently hard-coded to id 2
- [x] remove test scripts: reload.sh, radio-test.sh
- [x] rebase some commits - the git log is disgusting
- [ ] should `CC1201.sendData()` wait for transmission to complete before returning?  This code i [... cut]
closed 2016-05-27 by justbuchanan