RoboJackets/robocup-software #676: Tune ball placement behaviors

justbuchanan, 2016-05-22:
Ball placement works fairly well in the simulator, but will need some tuning in order to work with actual robots.  It could also use a little tuning to be more effective in the simulator.  See PR https://github.com/RoboJackets/robocup-software/pull/651

chachmu, 2016-07-14:
When I actually get around to doing this I will also try to make the play handle getting another stop command in a way that doesn't completely break everything.

jgkamat, 2016-10-02:
Can this be closed or do we need another round of this? (I don't know if ball placement is a challenge again?)

chachmu, 2016-10-02:
If we ever want to actually use this we will definitely need another round of tuning but if it doesn't come back up I doubt we would ever use it since the only way to activate it is to have both the external referee and the remote terminal open.

PR #987 (2017-07-23): Update ball placement to work with new rules
- [x] Move out of the way of ball placement for other team
  - [x]  Move out of the way of own ball placement?
- [ ] Fix coordinate system to match SSL standard
- [ ] Increase speed of placement
- [x] Fix issue where robot goes to random point outside of circle for a second after placing, sometimes running into the ball

Closes #985 and Closes #676.
PR #1001 (2017-08-25): Merging Changes from Competition 2017
Let's get this started.

Common: https://github.com/RoboJackets/robocup-common/pull/15
Firmware: https://github.com/RoboJackets/robocup-firmware/pull/40

Blocking #911 

Closes #985, closes #676
closed 2017-08-25 by joshhting