RoboJackets/robocup-software #686: MBED crashes and reboots itself periodically

justbuchanan, 2016-05-27:
It might be that it's running out of memory, but we're not quite sure.

cc @jjones646

jjones646, 2016-05-27:
Almost certainly a memory issue. I've been running everything through soccer for 15+ min now using ca04d6f, and it's been doing great.

jjones646, 2016-05-29:
This line in Console.cpp definitely isn't helping the issue...

```
// reserve space for 5 lines in rx buffer
_rxBuffer.reserve(400);
```

This seems to be heap allocation.

jjones646, 2016-05-29:
Another thing to note is that `_rxBuffer` is a `std::string` object. And it looks like under allocation is fine since all additions use `_rxBuffer.push_back`. So a reservation of 10 or 15 is more reasonable.

justbuchanan, 2016-06-13:
This file in the mbed repo has some info on stack sizes for the main thread and timer thread: https://github.com/mbedmicro/mbed/blob/master/libraries/doc/rtos.txt.

We would have to fork the mbed repo, but it may be worth looking into reducing both of those threads' sizes - main is definitely a lot bigger than it needs to be since most of our stuff happens in other threads.

justbuchanan, 2016-06-14:
Mostly fixed in #707

closed 2016-06-14 by justbuchanan