RoboJackets/robocup-software #696: circleci/ubuntu-setup is broken

justbuchanan, 2016-06-09:
It looks like an issue with the gcc package for firmware.  I created the `justin/ci-test` branch to track down the issue - it's the same as master except with an extra line appended to `ubuntu-setup` to trigger a rebuild of the CI image.  Here's the error:

~~~{.sh}
WARNING: The following packages cannot be authenticated!
  gcc-arm-none-eabi
E: There are problems and -y was used without --force-yes
[ERR] Your setup command FAILED. Exiting...
/home/ubuntu/robocup-software/external/DoCIF/util/maketest.sh: line 90: cd: /home/developer/project: No such file or directory

./external/DoCIF/commands/buildbaseimage.sh returned exit code 1

Action failed: ./external/DoCIF/commands/buildbaseimage.sh
~~~

jgkamat, 2016-06-09:
I think terry guo has decided to exact his/her revenge :/

Their repo is now private. They pulled a leftpad on us.

justbuchanan, 2016-06-09:
Dammit Terry...  I guess we need to find a new repo :package:

jgkamat, 2016-06-09:
In the interim, we can pull the one in the debian repos :wink: 

I'll see if that works, give me a sec

jgkamat, 2016-06-09:
Hmm, it seemed to complain about missing files (even though the headers actually exist). I think this is because I'm mixing trusty/yakkety dependencies. I'll try again properly sometime tomorrow.

justbuchanan, 2016-06-09:
Ok sounds good.  Thanks for looking into this!

jgkamat, 2016-06-09:
I think terry guo is a wizard and made a custom build that would work on 14.04, anything that I do dosen't seem to work 100%

I think the best course of action right now is to switch the CI to 16.04 (which builds at least the main build), since we need to do that later eventually. If anyone has the deb files that were distributed by terry guo, that would work too...

jgkamat, 2016-06-09:
https://github.com/RoboJackets/robocup-software/tree/jay/16.04: the 16.04 build works fine. Still would be nice to stay on 14.04 until after comp though

justbuchanan, 2016-06-11:
Haha yeah I messed around with some alternative gcc arm packages too and I can't get anything to work.

Switching to 16.04 is probably the easiest course of action since we're planning to do that after comp anyways.  It's fine by me if you want to go that route.

jgkamat, 2016-06-12:
Turns out the repo didnt get deleted, its just moved over here: https://launchpad.net/~team-gcc-arm-embedded/+archive/ubuntu/ppa

I'll add a PR to fix this hopefully.

jgkamat, 2016-06-12:
https://circle-artifacts.com/gh/RoboJackets/robocup-software/2204/artifacts/0/tmp/circle-artifacts.XWN3nrY/compile.txt

It seems like this newer version is causing the same errors I've been seeing. Any ideas? :/

justbuchanan, 2016-06-12:
Darn :/.  There's probably a way to fix it, but if the 16.04 version works fine, I'd just go with that.

jgkamat, 2016-06-17:
@ryanstrat needs 14.04 support, so i'm reopening this

jgkamat, 2016-06-17:
Going to start dumping things for myself here:
https://github.com/platformio/platformio/issues/301

PR #701 (2016-06-12): Jay/16.04
Closes #696

This switches the CI to 16.04, only minor changes were needed (most of this is removing hacks we had set up for 14.04).
PR #718 (2016-06-20): Fix 14.04 support
Closes #696

@ryanstrat could you check out the jay/14.04 branch on your computer and rerun ubuntu setup, and see if that works? 

If you've prevously run ubuntu-setup (on an older commit) could you run `sudo apt-get remove gcc-arm-none-eabi` before trying again?

I _think_ this works on a fresh install of 14.04, according to docker. I'm not going to change the CI because I'm too lazy, and this is way less hacky.

It dosen't seem to be working on the firmware build due to missing `strtok_r`? @justbuchanan do you know anything about this?

`#include <string.h>` dosen't fix it. [... cut]
closed 2016-06-20 by jgkamat