RoboJackets/robocup-software #706: make robot2015-gdb doesn't work

justbuchanan, 2016-06-13:
Running it gives me the error below.  It seems like the pyocd-gdbserver program isn't working right and gdb times out trying to connect to it.  I tried upgrading to the latest version of pyocd, but it still doesn't work.

~~~{.sh}
[100%] objcopying to make mbed-compatible executable
[100%] Built target robot2015
GNU gdb (GDB) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from build/firmware/firmware/robot2015/src-ctrl/robot2015_elf...done.
localhost:3333: Connection timed out.
You can't do that when your target is `exec'
Temporary breakpoint 1 at 0x7378: file /home/justin/src/rj/robocup-software/firmware/robot2015/src-ctrl/main.cpp, line 48.
The program is not being run.
(gdb)
~~~

justbuchanan, 2016-06-13:
It may be related to this issue: https://github.com/mbedmicro/pyOCD/issues/241

Sometimes pyOCD errors out with "The device has no langid"

jgkamat, 2016-08-29:
Closed in favor of rj/robocup-firmware's issue, linked above.

closed 2016-08-29 by jgkamat