RoboJackets/robocup-software #710: Log referee data

justbuchanan, 2016-06-15:
Right now the [Raw_referee](https://github.com/RoboJackets/robocup-software/blob/master/common/protobuf/LogFrame.proto#L91) field of the LogFrame is a `bytes` field, but it would be easier to use if it was an embedded protobuf mesage, similar to how we store vision data.

Also, the [simple_logger](https://github.com/RoboJackets/robocup-software/blob/09a5e6bddb1eb7aaa0a2bb8d920c972f3035f960/logging/simple_logger.cpp#L112) currently writes to this field, but soccer does not.  The place to add this to soccer would be in Processor.cpp.

If you could get this working on a new branch and send a pull request, that would be awesome!

ryanstrat, 2016-06-15:
If I change the bytes to a message, It'll break all previous log files again. Alternatively, I can add the message as a new field and we can have the parser see which one data is in.

justbuchanan, 2016-06-15:
Sounds good to me.  You can just put a comment next to that field noting that it's deprecated and add a new one for the protobuf.

I think the reason why that field is straight data rather than protobuf is because the legacy referee packets weren't protobuf.

ryanstrat, 2016-06-16:
There exists a complication with logging ref packets.

Best I can tell, we receive the ref packets in NewRefereeModule.cpp in the run() method, which is run async from the rest of the program.

A getPackets method exists, but it is not used. Instead, the changes to game state are pushed into the _state variable immediately upon arrival after parsing. The packet is then discarded.

Since the newly created raw_refbox field in the protobuf is a repeated field, it doesn't really matter how many packets we receive between main run loops, but I'm afraid of trying to add packets to the logFrame asyncronously. The current implementation of logging is not threadsafe.

Here are some solutions I came up with, but have not implemented:

1. Add a mutex lock for LogFrame to ensure that we don't write the logFrame while adding ref data in the other thread.
2. Process packets in the main control loop (either a new method or add it into the updateGameState method. Take advantage of the getPackets method. the run method in NewRefereeModule.cpp would just read in packets and store them in the circular buffer. It would be very similar to how vision is implemented.

@justbuchanan can you offer some insight please?

justbuchanan, 2016-06-16:
I think option 2 you listed is probably the best.  It makes sense for the Processor to do the work there since it does a similar thing for the the VisionReciver.  One thing to keep in mind is that we really only care about the latest vision packet - anything other than the most recent one can be discarded even if it hasn't been processed yet.

ryanstrat, 2016-06-16:
Do we care about old Ref packets either?

Will it ever transition into and out of a game state in less than one control loop? I can't think so under normal conditions.

ryanstrat, 2016-06-16:
I've kinda found a hybrid way. We're still parsing the ref packets immediately upon arrival. The packet then goes into a vector, just like vision packets, and are logged in processor, just like vision packets.

justbuchanan, 2016-06-16:
Ok that sounds pretty good to me.  It is possible (although pretty unlikely) that it transitions through two states in one control loop, although I don't think it would really hurt anything if it did.  Everything should be setup to care about the current state, not where it transitioned from.

ryanstrat, 2016-06-16:
In case you're interested, We receive a ref packet on about 2/3rds of runs through the loop.

I think it's logging them, but I'm having trouble getting the data out. I'll pick it back up in the morning.

I went ahead and pushed what I  have so far in 34be196

ryanstrat, 2016-06-21:
Closed with PR #717

closed 2016-06-21 by ryanstrat