RoboJackets/robocup-software #712: LineUp play sends the robot off-field

justbuchanan, 2016-06-16:
When running soccer with ssl-vision in the shop, the LineUp play tried to send the robot off of the side of the field.  See the picture below - the path's end is out of bounds.  Haven't looked into it too much yet.  Maybe something to do with the field geometry being wrong?

[image]

cc @jgkamat

jgkamat, 2016-06-16:
This might be related to the vision packet resizing I made? But those changes should show up in the window too. 

I cant reproduce this in the simulator, is this on the real field?

jgkamat, 2016-06-16:
[image]

The coordinates on your screen are the same as mine, so maybe the geometry resizing made the field smaller?

jgkamat, 2016-06-16:
If you comment out this line, does it work?
https://github.com/RoboJackets/robocup-software/blob/master/soccer/Processor.cpp#L305

I don't think theres a way to run the simulator on the same port as the vision (so I cant try a resize and then run the simulator)

commit 5d81f49d (2016-06-25): fixed field size issues. closes #712. see #566
closed 2016-06-25 by justbuchanan