RoboJackets/robocup-software #713: Seg Fault caused by Motor Status

ryanstrat, 2016-06-16:
Not sure if related to #674.

Here's the stack trace from GDB.

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff6a5934a in Packet::RadioRx::motor_status (this=0x7fffffffd500, index=0) at common/protobuf/RadioRx.pb.h:621
621	  return static_cast< ::Packet::MotorStatus >(motor_status_.Get(index));
(gdb) bt
#0  0x00007ffff6a5934a in Packet::RadioRx::motor_status (this=0x7fffffffd500, index=0) at common/protobuf/RadioRx.pb.h:621
#1  0x00007ffff6a53c42 in MainWindow::updateViews (this=0x9102a0) at ../soccer/MainWindow.cpp:592
#2  0x00007ffff6b0ecb5 in MainWindow::qt_static_metacall (_o=0x9102a0, _c=QMetaObject::InvokeMetaMethod, _id=2, _a=0x7fffffffd690)
    at soccer/moc_MainWindow.cpp:315
#3  0x00007ffff719a2a6 in QMetaObject::activate(QObject*, int, int, void**) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#4  0x00007ffff71a6622 in QTimer::timerEvent(QTimerEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#5  0x00007ffff719b054 in QObject::event(QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#6  0x00007ffff76ebc8c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#7  0x00007ffff76f0e56 in QApplication::notify(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#8  0x00007ffff7172c2d in QCoreApplication::notifyInternal(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#9  0x00007ffff71bf1ad in QTimerInfoList::activateTimers() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#10 0x00007ffff71bf661 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#11 0x00007ffff440be04 in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#12 0x00007ffff440c048 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#13 0x00007ffff440c0ec in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#14 0x00007ffff71bf98c in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#15 0x00007ffff717196b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#16 0x00007ffff71780e1 in QCoreApplication::exec() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#17 0x0000000000405d93 in main (argc=2, argv=0x7fffffffdfb8) at ../soccer/main.cpp:196

This was running in the simulator. Just opened the sim and left it for several minutes. Didn't actually press any buttons yet.

justbuchanan, 2016-06-16:
Interesting... I don't think I've seen this before.  I'll take a look at it.  Is there a way to reproduce this reliably?

ryanstrat, 2016-06-16:
I haven't gotten it to happen since, but I also haven't spent the time to
try. I was just lucky that I was debugging an unrelated segfault at the
time and was using gdb

Ryan Strat
ryan@str.at
(678) 517-3393
RoboJackets | President
Georgia Institute of Technology

jgkamat, 2016-10-02:
Not sure if this can be closed or not, can someone with more info on this topic comment on it?

jgkamat, 2017-07-01:
Closing due to timeout. It's likely this issue is resolved.

closed 2017-07-01 by jgkamat