RoboJackets/robocup-software #715: Goalie rampages if ball is placed near the edge of the goal.

chachmu, 2016-06-17:
Sometimes if the ball is placed behind the defenders but in front or to the side of the goalie pathplanning or something related freaks out and the robots start driving randomly in loops. It seems like the easiest way to recreate this is by placing the ball just in front of the corner of the goal. It doesn't seem to be related entirely to the normal path planning because it happens during both move and worldvelocity commands.

[logs.zip](https://github.com/RoboJackets/robocup-software/files/319738/logs.zip)

jgkamat, 2016-06-17:
Can reproduce this :+1: 

Seems to be an issue whenever the goalie goes into the clear state
Maybe it thinks it can't be in the box (its a pivotkick) and tries to leave, while simultanoulsy trying to clear the ball? Thats my first guess.

justbuchanan, 2016-06-19:
@ashaw596 any thoughts on this?

ashaw596, 2016-06-19:
My first thought is I think it might have something to do with the interactions with the giant shoot obstacle that gets created during the pivot kick. It seems to cause constant replanning at the moment and lag everything out.

justbuchanan, 2016-06-19:
It's possible, but the shoot obstacle doesn't apply to the robot doing the kicking.  I wouldn't think it would have an effect on the goalie in this case.

ashaw596, 2016-06-19:
Well I guess I'm having some trouble reproducing it when I'm just running the goalie then. When I run the plays everything lags out so  I can't really tell whats going on. When I run just the goalie, I think I ignore obstacles during pivot right now so it sometimes hits the wall and gets a bit stuck. (and sometimes almost pivots the ball into the goal and gets stuck) but I don't really see the circling? (except when the other robots are in the shoot obstacle and lags out the pathplanning)

justbuchanan, 2016-06-20:
I think when this happened before, it was when we ran the `TestDefense` play.  It took several tries to make it happen, but the goalie definitely went out of control - sort of a spiraling motion.

chachmu, 2016-06-23:
@ashaw596 I went ahead and used git bisect to try to find when the problem was introduce. The first time I did it I got that the issue was introduced in 84a40f5 which didn't really seem right, but after I tried again I got that the issue was in 022225 which also really doesn't make sense because that just merged master into the dynamic hit branch. Long story short, I can't really find exactly which commit it came from since its hard to tell the difference between the goalie rampaging and the computer lagging, but it does seem like it was introduced somewhere in #663.

ashaw596, 2016-06-23:
Yeah, if you disable the localobstacle in kick, it doesn't happen. I know its a problem in that the path is getting invalidated constantly since we are planning into an "obstacle". Its introduced by that commit since it increased the amount of tries it plans. I'll work on it when I get a chance.

ashaw596, 2016-06-23:
Ok, I "Fixed it". I had increased the number of times to retry pathplanning to 10 since dynamic obstacles can take another time. However, when the ball goes into the goal, the defenders try to get to the other side of the ball which now has a gigantic obstacle in the way so RRT planning fails. Apparently 20 failling maximum iteration Paths lags out soccer. I changed it to give up after one. We should probably stop trying to plan impossible paths when we can later.

PR #734 (2016-06-23): Albert/path planning fail elegant
Give up when RRTPlanning Fails.
Fixes #715
closed 2016-06-23 by ashaw596