RoboJackets/robocup-software #719: Coordinated_pass freezes occasionally

jgkamat, 2016-06-18:
When I run coordinated pass, it sometimes freezes, with the kicker kicking, but not actually moving towards the ball (without contact). I don't know if this is an issue in path planning or with plays.
[image]

chachmu, 2016-06-19:
I'm guessing that's path planning because I have been having that same issue for a while with multiple different plays.

justbuchanan, 2016-06-19:
@ashaw596 can you look into this and see if it might be path-planning-related?

ashaw596, 2016-06-19:
Its probably cause, I removed the requirement to not lose contact with the ball during pivot kick. I'll work on it.

ashaw596, 2016-06-20:
I couldn't really get reproduce the bug on my computer, but I changed aim to fumble if it loses contact again. It should fix it (theoretically) and doesn't seem to cause much issues.

PR #723 (2016-06-20): Change it to detect any loss of control of the ball as a fumble
Fixes #719.
closed 2016-06-20 by justbuchanan