RoboJackets/robocup-software #724: Update Simple-Logger to Save New Referee Data

ryanstrat, 2016-06-20:
The simple-logger program still records data by reading in the legacy referee data from the network.

The simple logger should store the newer protocol buffer based referee data.

It would also be nice to pull ball and robot information out of the vision packets to fit into the other fields of LogFrame protobuf so the logs are more similar to soccer logs.

ryanstrat, 2016-06-20:
I'm also going to add the ability for the simple logger to log from the simulator via a -sim flag.

I can't test the output, if I can't receive vision packets.

ryanstrat, 2016-06-21:
The branch for this won't compile until I merge the changes in for updating soccer logging, as the protobuf file is different.

ryanstrat, 2016-06-21:
Locally, I'm at a point where I'm logging ref packets and vision info from the simulator.

To develop this further, I want to parse the packets to fill the logFrames out like soccer does to make simple_logger logs compatible with the log_viewer and the log_converter.

I think I need to filter the vision data the same way soccer does it to account for camera overlap and such.

justbuchanan, 2016-06-22:
Hmm, I'm not entirely sure what the best thing is to do for the filtered vision data.  We have code that does that, but I don't know how easy it would be to use it in simple logger without copying a bunch of code from soccer or doing a decent amount of refactoring.

One option would be to just drop simple_logger entirely and just use soccer for logging.  Thoughts?

cc @barulicm

barulicm, 2016-06-22:
Logging with soccer is probably the easiest solution for near-term work. Best long-term solution would be to refactor to support code sharing between the two. I don't remember off the top of my head how difficult it would actually be to tear that part of soccer out into a separate library.

If we got any kind of algorithm working on our log data, it would be really cool to put together small boxes we could drop off at each field to just continuously log matches throughout comp. This would be easiest if Simple Logger existed as to minimize dependencies that would have to install on what would likely be a <$30 computer.

ryanstrat, 2016-06-22:
The filters we run the data through seem pretty embedded into soccer. I agree that yanking it out would be ideal.

In the short term, perhaps we could add an -observe flag or something that makes it so Soccer won't try to control the robots on the field? I can test how easy that would be today.

It's also, in theory, possible to post-process log files to filter the data.

ryanstrat, 2016-06-22:
After more study, it seems like the heavy lifting is done in /modeling, but it seems that over time, bits and pieces have been added on throughout soccer. Notably, the code to remove balls and robots on the excluded half of the field was placed in Processor.cpp (line 321 or so)

If we still need the ability to exclude half the field, which is set in the constructor of Processor, then we should refactor that code into the model files. However, I think we can get rid of that feature, since we're back to using just one size field.
@justbuchanan can you confirm that I can get rid of this feature?

barulicm, 2016-06-22:
@ryanstrat removing half the field is still very useful when practicing at comp while sharing the field with another team

ryanstrat, 2016-06-22:
Right.

I'll see if I can refactor that somehow.

I also feel like that should be selected via the GUI rather than a constant in the code

barulicm, 2016-06-22:
You can change it in the menus of soccer

ryanstrat, 2016-06-22:
Oh, I see how it's set up.

ryanstrat, 2016-06-22:
As long as we don't need the filtering for the simple logger, then we should actually be fine.

ryanstrat, 2016-06-22:
Here's the plan for the ball. The plan for robots would look similar.

Create and instance of `BallTracker` and `SystemState` in simple_logger.cpp.

I'd create a vector of detectionFrames and translate it to meters and with `_worldToRobots`, just like in `Processor::runModels()`. I'd skip the code to filter out portions of the field as we're only interested in logging matches. I pass the vector of `detectionFrames` into BallTracker::run(), which will perform filtering and store the result in a `SystemState` passed to it.

We can then read the filtered ball position from the instance of `SystemState`

The only code that would end up being duplicated is the transformation. All the filter code is isolated into separate classes. Should the modeling folder be moved to common?

justbuchanan, 2016-06-22:
To be honest, it may be easier to just use soccer for logging?  Is there any reason soccer won't work for what you want to do?  By default, soccer starts up with no plays selected to run and won't do anything other than log what it receives from vision/ref.

ryanstrat, 2016-06-23:
My main concern with soccer is that it will log extra debug data, which would not be applicable.

We could add a flag.

Long term, it would be nice to have a solution with fewer dependencies that requires less computing power. For this year, that will likely work.

JNeiger, 2019-01-20:
Closing due to simple-logger replacement in soccer

closed 2019-01-20 by JNeiger