RoboJackets/robocup-software #754: Soccer should look for the "logs" directory relative to the "run" directory

justbuchanan, 2016-06-30:
Right now it looks relative the current *working directory*.  This is an issue because if you launch it with `./run/soccer` from the root of the repo, it looks for `./logs` which probably doesn't exist.  This is why we failed to log our first match at comp.

chachmu, 2016-07-22:
This also seems to apply to the option to create a log folder under debug, I think it uses the same variable though so changing that should fix it.

jpfeltracco, 2016-08-31:
[Relevant Code](https://github.com/RoboJackets/robocup-software/blob/6d6947842789e90c67462b7fcbe8a21d8951764c/soccer/main.cpp#L157)

jgkamat, 2016-11-27:
Currently, the makefile cd's inside the run folder before running soccer when using `make run`, so anyone working on this, just change the logs dir in the two places in the referenced code, and test it out by manually running `./run/soccer`. I'll add the makefile changes once a PR is made, unless the assignee wants to stab at that too.

rachitk, 2016-11-30:
Just made a PR #835 to make the change requested. Not positive if it's the correct change or if it's fully complete yet, as `make run-sim` does not log properly with this change, likely due to makefile changes still needing to be made.

PR #848 (2017-01-15): Refactor refrences to be relative to soccer (in the run directory)
This fixes the various hardcoded file paths (relative to the current directory) to instead be relative to the directory where soccer is located, which makes it consistent. 

Closes #831, closes #754, closes #764. Replaces #835

Let me know if you find any other file paths referring to the current directory! Feedback is appreciated. :smile:
closed 2017-01-15 by jgkamat