RoboJackets/robocup-software #755: Restarting single robot behaviors isn't always the best thing to do

justbuchanan, 2016-06-30:
In coordinated pass, if the receiver is temporarily assigned to a different robot (which happens occasionally), the receiver gets restarted, which messes up coordinated pass transitions, causing the pass to hang indefinitely.

Restarting a single robot behavior when a new robot is re-assigned is often the correct thing to do, but not always.  This needs some thought.

jgkamat, 2016-07-14:
Could we just swap out the robot for a singleRobotBehavior without restarting it (if requested by the behavior, probably in its constructor). This is pretty important imo, especially for pass receives.

chachmu, 2016-07-26:
We could give behaviors a bool that would determine if they should restart on being transferred to a different robot then just check it to see if we should restart here in [SingleRobotCompositeBehavior](https://github.com/RoboJackets/robocup-software/blob/c89da1b7f75e8b28e0aee0cff80630db115ba02b/soccer/gameplay/single_robot_composite_behavior.py#L58).

jcarn, 2017-01-15:
I'll look into this and #756

jcarn, 2017-01-15:
Is there a specific circumstance in which this happens? So far its handling everything fine, I can't get it to hang

PR #897 (2017-03-05): Stopped CoordinatedPass from restarting behaviors when robots are switched mid-pass
Closes #755
closed 2017-03-05 by jgkamat