RoboJackets/robocup-software #756: Coordinated pass's receiver should always look for a robot close to the receive point

justbuchanan, 2016-06-30:
Relevant to #755.  The role requirements for the receiver's subbehaviors should be overridden to ensure that the assignment system looks for a robot close to the receive point, not the current ball position.

jcarn, 2017-01-15:
I'll look into this and #755

jgkamat, 2017-01-15:
@jcarn this issue *might* be fixed already (or will be fixed by #755). Someone needs to double check that though.

jgkamat, 2017-03-03:
This is *not* fixed yet, but #897 gets us most of the way there!

Right now, pass_receive overrides role requirements to prefer the robot closest to where the selected robot is going, which works, but prevents us from using the best robot to receive the ball. Ideally, we would delegate this out to capture, but it's role requirements don't take the ball velocity into account (so it assigns the kicker to get the ball, which fails due to the pass_receive's failure checking rules.

We need to either:

Change pass_receive's role requirements to prefer the robot which actually has the best chance of getting the ball
Fix capture's role assignment to consider ball velocity.
I would much prefer (2), but I'm not sure about how feasible that would be.

I would /really/ like to have this done by japan, and I think this is a pretty good issue for someone new who's interested in role requirements to look at!

jcarn, 2017-03-03:
I'm messing around with role requirements anyway for the stopped pass thing, so I'll keep working at it

jgkamat, 2017-03-05:
this is basically kind of a dupe of #375. Solving #375 will solve this issue too though.

kylestach, 2019-08-22:
@JNeiger, was this fixed by #1180?

JNeiger, 2019-08-22:
Yep, I think it was. Both due to the capture rewrite as well as the vision rewrite

closed 2019-08-22 by JNeiger