RoboJackets/robocup-software #758: Field edge obstacles are too big during corner/free kick plays

justbuchanan, 2016-06-30:
The obstacles we have make sense during normal play, but shouldn't apply when the kicker is setting up for a kick-in.

jgkamat, 2016-07-02:
Turns out this is an issue with the ball obstacle instead during setup in combination with our old link kick. I think albert's line_kick patch

jgkamat, 2017-03-14:
This is (in my opinion) our biggest problem when competing last year. We need to:

1. Evaluate if this issue exists
2. Fix it if it does

Hopefully it is fixed in the numerous line kick fixes made recently, and we can just close this. We'll probably need to adjust the field borders to test this since the field is way to small to support proper corner kicks like in comp.

jgkamat, 2017-05-08:
I don't think this issue is present any more, at least in the simulator. We can't close this until someone tries this on a full field though.

jgkamat, 2017-08-08:
we had no issues with field obstacles at nagoya, closing.

closed 2017-08-08 by jgkamat