RoboJackets/robocup-software #760: Config widget in soccer should have a wider first column

justbuchanan, 2016-06-30:
The default size is too small, so it has to be dragged wider to see the config variable's name.

chachmu, 2016-07-12:
According to Matt, the default column widths with QT creator don't always work properly. I tried setting the default width of the columns to 200 which works inside the QT creator display, but when I actually compiled soccer it was the same as before. One alternative is to make the minimum size for the columns be 200 which works but now you can't shrink the Name column any smaller than that. The Name column can still be made larger because it just pushes Value off the screen though.

chachmu, 2016-07-12:
As a side note, 175 seems like a nice width because it splits the widget exactly in half between Name and Value.

PR #779 (2016-08-21): Increase size of Name tab for Config Widget
Make minimum size of Name tab 175. Closes #760.
closed 2016-08-21 by jgkamat