RoboJackets/robocup-software #764: Playbook load in soccer should show file chooser in the correct folder

justbuchanan, 2016-06-30:
Right now it shows the root of the project, it should launch to `soccer/gameplay/playbooks`.

jpfeltracco, 2016-08-31:
[Relevant function](https://github.com/RoboJackets/robocup-software/blob/a10654f07fdf73f2951ae9a0081b865a23102880/soccer/gameplay/playbook.py#L7)

[Relevant call to function](https://github.com/RoboJackets/robocup-software/blob/a10654f07fdf73f2951ae9a0081b865a23102880/soccer/gameplay/main.py#L147)

jgkamat, 2016-11-27:
So the reference above is for the actual loading of the file, for the file chooser you will need to look at [this bit of code](https://github.com/RoboJackets/robocup-software/blob/6ea6cf696dd5ecd7caca044bee93a308cea32e5c/soccer/MainWindow.cpp#L1203). However, this already works when running soccer from the run directory. See #754 as well.

I think all paths should be relative to the root of robocup-software, so I'm in favor of changing the `../soccer/gameplay/playbooks/` to `./soccer/gameplay/playbooks/`, and swapping it over in the makefile. If you want to tackle this issue, go ahead and do the above for now.

This relative/absolute path is starting to get kind of annoying, so I'm going to open another issue for it.

rachitk, 2016-11-30:
Made the change in #835 as requested; no changes have been made to makefile yet, so `make run` and the like now do not work while `./soccer/run` does.

PR #848 (2017-01-15): Refactor refrences to be relative to soccer (in the run directory)
This fixes the various hardcoded file paths (relative to the current directory) to instead be relative to the directory where soccer is located, which makes it consistent. 

Closes #831, closes #754, closes #764. Replaces #835

Let me know if you find any other file paths referring to the current directory! Feedback is appreciated. :smile:
closed 2017-01-15 by jgkamat