RoboJackets/robocup-software #768: Project Overhaul

guyfleeman, 2016-07-04:
Split the repository into Common, Software, and Firmware
- [ ] asd

Fix Labels

justbuchanan, 2016-07-06:
Keep in mind that splitting the repo does add some overhead when modifying shared components.  For example, if you modify something related to the radio protocol definition, you'll have to:
* commit in robocup-common
* submodule update and commit in robocup-software
* submodule update and commit in robocup-firmware

Whereas previously this would have just been one commit.  I'm not saying splitting the repo is a bad move, just be sure to weigh the pros and cons.

jgkamat, 2016-07-06:
Yah its going to be a bit harder to do these kind of things, but it will be way neater for the new guys (hopefully) without firmware getting in the way (unless we want it to).

I'm going to see if I can add submodule update hooks for checkouts, (since we've had that as a problem for a while), Since common is a submodule, we can commit it on firmware (or soccer), test it with the new changes, then commit common+soccer (common as a submodule, then push) and then we would have to deal with firmware.

Its pretty easy to reintegrate it if we have issues, its way harder to seperate them out than put them together. The biggest issue that I see is teaching everyone about submodules for real this time, but we really should get to that anway (since we use them extensively already). 

Hopefully this will stick, but if it dosen't that shouldn't be a disaster either

jgkamat, 2016-08-21:
This is mostly done, barring readme changes and issue migration. There are only issues labeled as 'firmware', so I think we can do this manually. @guyfleeman, are there more issues you wanted to migrate to robocup-firmware other than the ones labeled as firmware currently?

guyfleeman, 2016-08-21:
No just the firmware ones. I found this
https://github.com/google/github-issue-mover
if you want to avoid doing it manually but I'm not sure what features it has, like filtering by label.

jgkamat, 2016-08-29:
Issues labeled as firmware have been moved for the most part. I left all the issues related to the radio protocol here though. In the future, I think we should either keep radio issues here, and break off significant firmware problems in rc-firmware.

jgkamat, 2016-08-31:
almost done, need to make a readme for common, as well as work on the docs site for firmware.

jpfeltracco, 2016-08-31:
I know we have "split and migrate the firmware packages" listed as checked off, but right now there is no difference between the arch-packages.txt in both repos (aside from me adding SDL2 to robocup-software's list), do we need a new issue for that?

jgkamat, 2016-09-01:
I've split the ubuntu packages, but I'm hesitant on splitting the arch ones, as I might remove something important...

I'd rather just keep everything, as we don't have many arch people, but I could comb through it and try to get that down...

jpfeltracco, 2016-09-01:
Yeah, that would probably just be wasted effort, I doubt someone in Electrical is really going to be running Arch and only want to build firmware and not install packages for soccer. We can probably just clean up as we go along and not worry about it too much.

jgkamat, 2016-10-02:
I'm going to call this done. *sigh*

Issues that come up will be deat with later.

closed 2016-10-02 by jgkamat