RoboJackets/robocup-software #770: Implement full angle paths with motion/rotation constraints

ashaw596, 2016-07-07:
I've noticed that fast turning causes Pathing errors even in simulator, and this should be implemented eventually anyways to have accurate shooting and stuff.

Currently, PID is used to control angle on the soccer side with a target Angle or Location in MotionControl.cpp. I've already modified Paths and Pathplanner to have the capability to have angle information, but haven't implemented actual angle path planning. InterpolatedPath has angles in their Waypoints.

We would use trapezoidal motion planning for rotation and stuff.

jpfeltracco, 2016-10-23:
These pathing errors show up in our simulator when running the `RepeatedTurningLineUp` play where the robots completely diverge from their paths because they are attempting to rotate while at full velocity.

This divergence can be minimized by capping the angular velocity by 
```cpp
rotationConstraints.maxSpeed - C * target.vel.mag()
```
Where C is some constant (0.9 seemed to work okay). This essentially prevents the robots from rotating while at max velocity and seems to mostly remove the pathing errors.

This branch has this hack in it: [jeremy/pathingHack](https://github.com/RoboJackets/robocup-software/tree/jeremy/pathingHack)

Full angle path planning that takes into consideration the full constraints of the robot is in the works.

ashaw596, 2016-10-24:
Nice!

jpfeltracco, 2016-11-24:
[This Issue is related](https://github.com/RoboJackets/robocup-software/issues/781).

So my rough idea for solving this is:
1. Add a max wheel speed constant.
2. Run the paths through an angle planning function after [here](https://github.com/RoboJackets/robocup-software/blob/master/soccer/planning/IndependentMultiRobotPathPlanner.cpp#L78)
3. The function would first pick an ideal angle rotation speed, then calculate the wheel speeds required to achieve both that rotation and velocity using the translation/rotation to wheel velocity matrix.
    * If any wheel speed exceeds the max wheel speed, we just cut the angle rotation speed in half and check again (or find a way to calculate the rotation required to bring us up to max wheel speed)
    * Otherwise, simply accept that angle speed.

To implement this I believe I would have to add a rotationCommand to the PlanRequest struct, but I'm a little confused with whether or not I should add the rotationCommand to the SinglePlanRequest as well, and why exactly we have these two similar structs.

Also, do you think it would be a better idea to do angle planning at the same time we're fitting the velocity curve? That way I guess we could scale back the translational velocity if we can't finish the rotation in time or something.

jpfeltracco, 2017-01-15:
Ignore what I wrote previously, this was not an issue with exceeding max wheel speed, instead the problem is that we were clamping acceleration in robot coordinates, not in field coordinates. This caused massive accelerations to be caused whenever we rotated. This will be fixed in #781. Full angle planning will eventually be done through this issue.

For reference, this is the paper we're using: http://www2.informatik.uni-freiburg.de/~lau/students/Sprunk2008.pdf

JNeiger, 2018-10-25:
Did the full angle stuff get implemented? 

Could you also specify what you mean by full angle planning? Is this just taking into account acceleration limits in addition to linear acceleration limits?

ashaw596, 2018-10-25:
Nope. We don't plan rotation together with motion. Main issue is we just sort of do trapezoidal motion planning right now. It's basically like a pid right now with the target rotation angle.

closed 2019-08-25 by kylestach