RoboJackets/robocup-software #772: Make soccer continue to run if goalie looses vision or is deselected

jgkamat, 2016-07-11:
Currently all robots halt in this case, which is unacceptable. This is only applicable when we are running defense.

chachmu, 2016-07-11:
Do the other robots stop if the goalie loses radio? I thought they did during comp but I could be wrong.

jgkamat, 2016-07-11:
yes, I'm fairly sure this is the case (the same effect as selecting a goalie that does not exist, its a failed role assignment loop)

chachmu, 2016-07-12:
Currently if there is no goalie selected the root play tries to assign the goalie to robot -1. This causes the role reqs system to error. I think part of the solution to this problem involves assigning an arbitrary goalie if the refbox is not sending a goalie id and soccer does not have a goalie selected. Should I just go ahead and make it so the goalie is set by:

1. Refbox packet
2. Soccer goalie selector
3. Arbitrary robot (maybe the lowest robot number on the field during startup?)

jgkamat, 2016-07-12:
(3) is a bad idea, as we would get penalized if that happened during a game (maybe we lost the ref for a while). Instead for 3, soccer should just sit tight and wait (no goalie runs) until we get the id back. 

If we did this, we would want to add an extra goalie textfield, so we can know which goalie soccer is actually assigning to.

Be sure to branch off of comp2016 if you do this though...

chachmu, 2016-07-12:
I could make it keep the same goalie id as previous if whatever value its being given is invalid.

jgkamat, 2016-07-12:
I still think 3 is a bad idea (we might not want a goalie sometimes, maybe only one testing robot). Instead it would be better to have the system be able to cope w/o a goalie.

chachmu, 2016-07-12:
Wouldn't you just use the handles goalie if you don't want a goalie?

jgkamat, 2016-07-12:
hmm, that toggles all of defense though...

I mainly don't like setting an arbitrary goalie because that's unpredictable (and we have to select a goalie at the start of a match).

chachmu, 2016-07-12:
Ok, I will look into it. I think I just figured out the problem.

chachmu, 2016-07-12:
So it turns out root play handles not having a goalie just fine, its defense.py that is causing all of the problems. I will see if I can easily make it handle not having a goalie.

jgkamat, 2016-07-12:
yup, its something to do with the role reqs in defense I think.

It was working last year (at the start of comp) and something broke it, I dont remember what. 

If you are stuck for a while I would check check out the defense from before the china comp and see the diff

chachmu, 2016-07-12:
I found the issue, is there a way to make role assignments just not assign a robot? Currently defense.py requires having a subbehavior called "goalie" which then causes the role reqs to fail when it tries to get a robot with an invalid shell id.

jgkamat, 2016-07-12:
hmm, I'm not sure... Maybe you could set the priority to be really high but set required to false? I think that would work.

chachmu, 2016-07-12:
I don't think that would help because it would still try to check the shell id which is where it throws the error. I guess I will see if I can modify defense to not require the goalie subbehavior to exist.

jgkamat, 2016-07-12:
Another thing that might work is we could disable the goalie if we can't see the said robot shell (I think thats possible), or we could modify role requirements to ignore shell id's <0

chachmu, 2016-07-12:
If this doesn't work I will probably end up making role requirements just not assign a  robot if you ask for a shell id that is None.

jgkamat, 2016-07-12:
@justbuchanan  :cry:

chachmu, 2016-07-12:
Ok, so technically I have it working although the code seems a little hacky since its mostly a bunch of if(goalie exists): added in. I can push this so you can look at it but I think it might be better to change the way role requirements handles invalid shell ids. Currently it fails if you try to assign a robot number that is not connected so just having it ignore None wouldn't work if the goalie lost radio. Maybe just have it not assign a robot to any subbehavior that requests an invalid shell id?

justbuchanan, 2016-07-12:
This may require a change to the role assignments system.  As is, if the goalie behavior is marked as `required`, the role assignment will fail if the robot with that shell id isn't available.  If you mark it as optional (`required=False`), it may assign the goalie robot to a different behavior because optional behaviors are assigned after required behaviors.  I'll think about it some more and let you know if I have any ideas for handling this cleanly.

chachmu, 2016-07-12:
I didn't change the goalie required flag to false so that has always been like that although it would probably be good to fix that. Should I try changing the role requirements system to see how that works before we decide which method to use?

chachmu, 2016-07-13:
I went ahead and tried changing role reqs to not assign robots to behaviors requiring an invalid shell id. If we make it so that python interprets no goalie assigned as shell id -1 like the c++ side does instead of None then it only takes like 3 lines changed to make it work.

chachmu, 2016-07-13:
@justbuchanan @jgkamat I am doing some work to make the goalie get updated from the autoref that will probably get rolled in with this stuff too. I thought it might be a good idea to disable the goalie id selector and set it to the right id if you are using an external ref so there isn't any confusion. If I do that, any time we try to open simulator we would have to either have refbox open or uncheck the use external ref option every time we want to test a play. Should I make soccer default to not using an external ref if you are using -sim? Or is there a better way to do this?

jgkamat, 2016-07-13:
Could we use the ref selector to override the goalie id selector in soccer? that would be the best way of doing this imo.

chachmu, 2016-07-13:
Thats what happens now although I have the goalie selector disabled when the external ref box is checked so its clear that you aren't the one setting that. I guess I could look into making it only gray out if the ref box is actually connected so that way if you start simulator without running refbox you can still choose the goalie yourself.

chachmu, 2016-07-13:
I went ahead and made a new branch with everything working where role reqs is modified instead of defense and it seems a lot cleaner. In order to let you guys take a look would it be better to make a new pull request alongside the old one or delete the old one first?

commit a998c460 (2017-01-11): Fixes #772 (Soccer runs without goalie)
PR #852 (2017-01-13): Final Goalie Fixes
I am recreating this pull request because the old one has more lines of merge errors than I am actually adding code. This pull request:

- [x] Closes #772
- [x] Closes #765
- [x] Closes #761 
- [x] Keeps the previous goalie selected if the external ref goalie id changes to be invalid
- [x] Disables the Ref Shortuct bar IFF an external ref is connected
- [x] Disables the goalie selector IFF the external ref is transmitting a valid goalie id
- [x] Add option for No Goalie
- [x] Grey out Invalid Goalie IDs in the dropdown menu

Continuation of #777 and #774 and #783
closed 2017-01-13 by jgkamat