RoboJackets/robocup-software #773: Remove simple-logger

ryanstrat, 2016-07-11:
Simple-logger should be deprecated in favor of the official [SSL-logtools](https://github.com/michael-bleier/ssl-logtools)

It currently isn't doing anything harmful except opening up the possibility of creating log files in yet another format.

jpfeltracco, 2016-08-31:
Updated link the the new [SSL-logtools](https://github.com/RoboCup-SSL/ssl-logtools).

rpaulucci3, 2017-01-11:
Is this still open?

jgkamat, 2017-01-11:
Yes, but I this issue needs to be properly triaged. I haven't used either logging tool (other than our log _viewer_ which is here: https://github.com/RoboJackets/robocup-software/blob/master/soccer/LogViewer.cpp). 

I think that the logging tools are meant to log the vision packets, not soccer (which means they don't log our soccer program, they log the state of the field caputred by the vision cameras). This could be as simple as deleting those files, but we want to be sure we don't have any logs in that format. 

Another (stretch and slightly off topic) goal is adding the ability for our log viewer to open the log files generated by the SSL logger, but this but I'm not sure if this issue should track that. Talk to me next meeting about this issue, but I think it should be fine to simply remove this file and it's build target.

ryanstrat, 2017-01-11:
So fundamentally, we can't get rid of it until we can actually handle reading the official log format.

I worked a lot on Logging over the summer, but most of it has not been merged to master and I did not have time to work on it last semester. Logging is currently challenging because our primary log format saves the filtered world model, rather than the raw vision inputs, which is helpful for a variety of reasons. The SSL log format is basically just writing the network packets into a protobuf (mostly). We currently do not have a way to properly read un-filtered log files within soccer. The best solution for handling the SSL log files might be to write a utility to rebroadcast the packets over the network and emulate ref/vision systems to soccer.

ryanstrat, 2017-01-11:
Also, at some point I was working on a log converter that could shift logs between SSL format, RJ format, and BSV format. That is partially working on one of my branches.

JNeiger, 2019-01-18:
@jgkamat We basically can read the official log format in soccer so we can remove simple logger now right?

Would close #724 too

closed 2019-01-20 by JNeiger