RoboJackets/robocup-software #776: Play KickGently runs during Kickoff Prep

ryanstrat, 2016-07-13:
Seems like it just needs to be removed from the playbook or altered to not run during kickoff prep.

chachmu, 2016-07-13:
I can fix that, originally we were going to use this play for the match against the robobulls but it wasn't ready in time to test it so the behavior is still set to always run.

jgkamat, 2016-07-13:
iirc, the kick_gently play has a score == 0, and thats causing it. Changing the score to something other than zero should fix this for us?

chachmu, 2016-07-13:
We changed the score to that because we thought we were going to use only this play during that match.

chachmu, 2016-07-14:
@ryanstrat Ok so there definitely is a problem with kick_gently which I will fix but I'm not able to reproduce it running during kickoff prep. Which playbook are you loading?

ryanstrat, 2016-07-14:
I was just loading the default. I can capture some logs of it, if you'd like

chachmu, 2016-07-16:
@ryanstrat I already fixed what was causing the issue on the comp2016 branch so it doesn't really matter.

closed 2016-08-02 by jgkamat